ros2-rust/ros2_rust #1: Build Error

ludusrusso, 2018-03-05:
Hi, 
when I try to compile it (in docker Ubuntu 16.04) I got this error:

```
+++ Building 'tinyxml_vendor'
==> '. /rust_ws/build_isolated/tinyxml_vendor/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/rust_ws/build_isolated/tinyxml_vendor'
==> '. /rust_ws/build_isolated/tinyxml_vendor/cmake__build.sh && /usr/bin/make -j3 -l3' in '/rust_ws/build_isolated/tinyxml_vendor'
[ 12%] Performing patch step for 'tinyxml-2.6.2'
No such file or directory
CMakeFiles/tinyxml-2.6.2.dir/build.make:101: recipe for target 'tinyxml-2.6.2/src/tinyxml-2.6.2-stamp/tinyxml-2.6.2-patch' failed
make[2]: *** [tinyxml-2.6.2/src/tinyxml-2.6.2-stamp/tinyxml-2.6.2-patch] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/tinyxml-2.6.2.dir/all' failed
make[1]: *** [CMakeFiles/tinyxml-2.6.2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Any help? I do not find some log to understand it!

esteve, 2018-03-08:
@ludusrusso thanks for trying out `ros2_rust`! I think this error is caused because `patch` is not installed on the system, could you install it (`apt install patch`) and try again?

ludusrusso, 2018-03-08:
Many thanks! I've solved it. I lost some dependencies from ros2! Sorry I'm quite new about Ros2 and rust so it was some trials!

Anyway, I like very much rust as a language (even if I know it since few weeks), I'll for sure will try the rclrs client!!

closed 2018-03-08 by ludusrusso