ros2-rust/ros2_rust #2: Status?

VTCAKAVSMoACE, 2018-09-17:
Last commit was 8 months ago. Is this still under development or should I fork this and attempt to continue it (poorly)?

esteve, 2018-11-28:
Sorry for the slow response, didn't see the ticket before.

I'd be very happy to accept patches, pull requests and ideas for improvements and new features. Let me know if you'd be interested in contributing.

Thanks.

tprk77, 2019-03-02:
Hi Esteve! Looks interesting, but it seems it doesn't work with Bouncy (or Crystal). :crying_cat_face:

I'm kind of curious on OSRF's stance on Rust. I'm assuming this is a side project for you?

esteve, 2019-03-03:
Hey Tim! I'm no longer at OSRF, but I don't believe a Rust client would make it into the official tree. This is just a side project, so I haven't had much time to flesh it out, but fortunately @lelongg has updated it for Crystal and made a whole bunch of improvements, including lots of new features in #3

nnmm, 2022-03-18:
I'm closing this issue since it is outdated.

closed 2022-03-18 by nnmm