ros2-rust/ros2_rust #5: message and crate generation

christianrauch, 2019-02-28:
I am trying to build https://github.com/esteve/ros2_rust/pull/3 but the examples fail because of the missing `std_msgs` crate. I manually cloned `common_interfaces` to the workspace and the `std_msgs` get successfully generated in `install/std_msgs/share/std_msgs/rust/src`.
I then added `std_msgs = { path = '[...]/rust_ws/install/std_msgs/share/std_msgs/rust' }` to the `dependencies` section in the `rclrs_examples` but cargo still cannot find the `std_msgs` crate.
I did not apply the patch `rosidl_typesupport_ros2_rust.patch` since the rust code was successfully generated.

How do I set the path to the crate such that cargo is able to build the examples?
Also, is it possible to reuse already installed message definitions (e.g. from package `ros-crystal-std-msgs`) instead of duplicating the messages in the workspace?

christianrauch, 2019-02-28:
The path to the `std_msgs` crate was already set and the compilation failed because I set it twice.

I can compile the examples now by registering the generator: https://github.com/lelongg/ros2_rust/pull/1

nnmm, 2022-03-18:
This should not be an issue anymore when following instructions in the README, and to the extent that it is still an issue, I think https://github.com/ros2-rust/ros2_rust/issues/11 covers it.

closed 2022-03-18 by nnmm