ros2-rust/ros2_rust #10: Outreach with the Rust Community

ruffsl, 2019-03-05:
It'd be nice to get some more feedback on the design/implementation of rclrs from the larger Rust community, e.g. are there better approaches to wrapping the rcl c library, code generation, testing practices, API pattern that ROS users could benefit from. Additionally, many Rust users may appreciate what the rest of the ROS2 ecosystem has to offer in terms of robotic software development, growing the number of users and subsequent contributions. As this project has remained somewhat inactive over the year given limited resources of the original author, what ways could be better facilitate greater maintenance for the project?

roboticswithjulia, 2025-03-20:
I presented at the RustMeet conference this weekend, showcasing custom application development with our crate. My key takeaway: robotics remains relatively unexplored within the Rust community, yet attendees showed significant interest, immediately brainstorming potential projects they could tackle independently.
The presentation generated substantial engagement:

Video available here: https://vimeo.com/event/4991254 (presentation at 3:04)
Q&A session with numerous questions (starting at 3:47)

This is the shape of the knowledge of people before the conference [image]  

Additionally, I conducted a hands-on workshop using this repository: https://github.com/roboticswithjulia/ros2_rust_workshop, which demonstrates a custom application approach for participants to get practical experience.
This could serve as valuable initial feedback from the Rust community. I connected with several Rust groups at the conference:

Rust Vienna (https://www.meetup.com/rust-vienna/)
Rust Gdańsk (https://www.meetup.com/pl-PL/rust-gdansk/)
RustLang Czech Republic (https://rustlang.cz/)

I also discovered BCN Rust (https://www.meetup.com/es-ES/bcnrust/). Perhaps we should consider broader engagement with the Rust community regarding our repository and its roadmap? We could publish a LinkedIn post to gather feedback or propose talks at these meetups.
What I learned about technical challenges: many Rust developers use MacOS, which created compatibility issues with Docker containers. I encountered numerous problems with Gazebo simulations—many couldn't open the simulator due to graphics card limitations. While some resolved this with simple xhost + commands, others faced persistent issues, particularly those with i5 processors where the Gazebo simulation simply wouldn't run. Given these obstacles, I'm considering switching to noVNC (https://novnc.com/info.html) as an alternative visualization solution.
@ruffsl - Would this align with what you mentioned earlier?

esteve, 2025-03-20:
@roboticswithjulia thanks for all interest in the project. However, nobody who's involved in the development of the project has given a talk yet because we know that it's still unstable and that would mislead users, making them feel frustrated when trying ros2-rust for their applications when things don't work as they expect.

It's important to manage their expectations regarding the project, I'm sorry to say it, but I watched your talk and some of the answers you gave during the Q&A were unfortunately not reflecting the reality of the project or just incorrect, and I hope it's not the case, but it might have misled users as to what to expect from ros2-rust. Right now we're being very careful when we discuss ros2-rust because it's still not stable and as it usually happens, the first impression is the one that matters the most.

In any case, I really appreciate your enthusiasm and I hope to see your first contribution soon.

This ticket is quite old (~6 years), we already have a design that we're following. I'll probably close this ticket sometime soon.

roboticswithjulia, 2025-03-20:
I truly understand your concern about managing expectations for the ros2-rust project. You're absolutely right that presenting unstable projects can lead to user frustration when things don't work as expected. I apologize if my talk and Q&A session contained information that didn't accurately reflect the project's current state.
I would greatly appreciate your feedback about which specific points were misleading or incorrect in my presentation. This would be invaluable for my own learning and to ensure I don't spread misinformation in the future.
Rest assured, this will be the last time I speak publicly about the project. While I had hoped that discussing it might generate useful feedback before completion, I completely respect your position as one of the creators. You naturally have a much deeper understanding of the project's status, design direction, and development roadmap.
Thank you for your kind words about my enthusiasm for the project. I'm looking forward to contributing in more constructive ways going forward.