ros2-rust/ros2_rust #11: Improve message generation

ruffsl, 2019-03-05:
Having to include and recompile all the message types any rust project requires is kind of a non-starter: 

> "Ain't Nobody Got Time for That"
> ~ Kimberly "Sweet Brown" Wilkins

Are there anyways we can improve upon this, like upstreaming `rosidl_generator_rs` as one of the default IDL generators, or reusing IDL files generated from `rosidl_generator_c`? How have other client libraries such as `rclpy` worked around this? Are there memory safety considerations we should account for? Already I've encountered memory leaks using rclrs that could have been prevented by rustc and RAII.

esteve, 2019-03-19:
> How have other client libraries such as rclpy worked around this?

I can't speak for other client libraries, but definitely can for rclpy. rclpy hasn't worked around this issue, it just happens to be included upstream, but there are no mechanisms in rclpy for not regenerating messages.
 
> Are there memory safety considerations we should account for? Already I've encountered memory leaks using rclrs that could have been prevented by rustc and RAII.

Could you file a ticket with an example so we can address this? Thanks.

nnmm, 2022-04-22:
> Having to include and recompile all the message types any rust project requires is kind of a non-starter

I'm not sure if we should try to do something about this (not sure if it's possible?), or simply hope we'll get included into ROS 2 proper and have `rosidl_generator_rs` become a default generator.

The other points in this issue have been addressed, I think.

nnmm, 2022-05-28:
@esteve what do you think about my previous comment? Can this issue be closed as "won't fix until rosidl_generator_rs is included in the ROS 2 distribution?"

esteve, 2025-03-20:
Closing issue, @nnmm is right that this won't be really fixed until the generator is included as one of the default message generators.

closed 2025-03-20 by esteve