ros2-rust/ros2_rust #12: Idiomatic API for Rust

ruffsl, 2019-03-05:
Rather than merely a wrapper around the rcl c library and types, what other ways can we make the rclrs API more Idiomatic to rust design patterns and style? How do we expect to expose error handling, logging, composition of nodes, etc. This is intentionally open ended, but please feel free to narrow in scope.

tprk77, 2019-03-16:
Can you give some specific examples of what you mean? It's not like the current API is using raw pointers and using `int` for error codes. It already seems pretty Rusty to me. For example, it's using structs, traits, results, etc.

esteve, 2019-03-18:
One improvement I didn't have time to make was to tie the lifetime of `Publisher`s and `Subscription`s to `Node`s, but beside that, I'd say the API is fairly idiomatic. Like @tprk77 asked, @ruffsl do you have a concrete example of where the API of `rclrs` is not idiomatic? Thanks.

closed 2025-03-20 by esteve