ros2-rust/ros2_rust #21: IDL Support (Dashing+ Support)

AndrewJSchoen, 2020-05-12:
I wanted to separate out this specific issue to prevent overloading the topic in the main Dashing support thread. Specifically, this is a discussion for the planned feature of IDL support, required by ROS2 Dashing and later versions. 

Given the discussions there, there seem to be a couple of issues that need resolving, one major one being the choice of either using static compilation or introspection. I would be interested in any opinions that people have about that, as I look myself into how best to go about this.

lelongg, 2020-05-20:
For reference here is the link to the start of this discussion: #19 

Looking at the state of RTPS and IDL support on rust, here is a pair of interesting developments: [rtps-gen](https://github.com/frehberg/rtps-gen) and [rtps-rs](https://github.com/Klapeyron/rtps-rs).

It doesn't look ready but using these could prevent having to copy messages from Rust to C before serialization (and vice versa for deserialization).

nnarain, 2020-06-12:
Hi. Just wanted to say I'm also interested in this and giving it a shot. Current changes on the [dashing-plus](https://github.com/ros2-rust/ros2_rust/compare/master...nnarain:dashing-plus) branch in my fork.

Ported the IDL generator and made some fixes for the rcl API. Publish and subscribe examples are working. Need to clean it up though.

Also, I'm specifically building against Dashing. Figured I'd do Eloquent and Foxy too (Not sure to what extent things need to change there).

nnarain, 2021-07-07:
Wanted to note that Galactic has introduced some new tools for `rosidl`. Might be easier now to integrate with `cargo`.

nnmm, 2022-03-18:
I think this can be closed now – after #80, IDL support should be practically complete, at least in Foxy+.

closed 2022-03-18 by nnmm