ros2-rust/ros2_rust #27: Array content not published

LoyVanBeek, 2020-09-26:
When publishing messages that consist of an (unspecified length) array, that array is not filled and always received empty. 

I'm busy making a ROS2 (Foxy) driver for an old Neato vacuum robot using https://github.com/LoyVanBeek/neato_rust. I just started learning Rust and wanted to do more ROS2, so why not cause myself a lot of trouble learning 2 things at the same time?

The Rust interface with the robot over serial works fine, next step is to wrap that driver in a ros2-rust Node. I can perfectly fine publish a message, but when publishing a `sensor_msgs/LaserScan`, there is no content for the `ranges` field. 

From what I gather, the generated code concerning a `LaserScan` message is this, but AFAICT, there is no code generated to handle the ranges/intensities. 
Same issue holds for eg. an Int8MultiArray (which I thought was the simplest message having an array)

```rust
#[derive(Default)]
pub struct LaserScan {
    pub header: std_msgs::msg::Header,
    pub angle_min: f32,
    pub angle_max: f32,
    pub angle_increment: f32,
    pub time_increment: f32,
    pub scan_time: f32,
    pub range_min: f32,
    pub range_max: f32,
    pub ranges: Vec<f32>,
    pub intensities: Vec<f32>,
}

#[link(name = "sensor_msgs__rosidl_typesupport_c__rsext")]
extern "C" {
    fn sensor_msgs_msg_laser_scan_get_type_support() -> uintptr_t;

    fn sensor_msgs_msg_laser_scan_get_native_message(
    angle_min: f32,
    angle_max: f32,
    angle_increment: f32,
    time_increment: f32,
    scan_time: f32,
    range_min: f32,
    range_max: f32,
    ) -> uintptr_t;

    fn sensor_msgs_msg_laser_scan_destroy_native_message(message_handle: uintptr_t) -> ();

    fn sensor_msgs_msg_laser_scan_angle_min_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_angle_max_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_angle_increment_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_time_increment_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_scan_time_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_range_min_read_handle(message_handle: uintptr_t) -> f32;
    fn sensor_msgs_msg_laser_scan_range_max_read_handle(message_handle: uintptr_t) -> f32;
}

impl LaserScan {
  fn get_native_message(&self) -> uintptr_t {
    return unsafe { sensor_msgs_msg_laser_scan_get_native_message(
    self.angle_min,
    self.angle_max,
    self.angle_increment,
    self.time_increment,
    self.scan_time,
    self.range_min,
    self.range_max,
    ) };
  }

  fn destroy_native_message(&self, message_handle: uintptr_t) -> () {
    unsafe {
      sensor_msgs_msg_laser_scan_destroy_native_message(message_handle);
    }
  }

  #[allow(unused_unsafe)]
  fn read_handle(&mut self, _message_handle: uintptr_t) -> () {
    unsafe {
      {
      self.angle_min = sensor_msgs_msg_laser_scan_angle_min_read_handle(_message_handle);
      self.angle_max = sensor_msgs_msg_laser_scan_angle_max_read_handle(_message_handle);
      self.angle_increment = sensor_msgs_msg_laser_scan_angle_increment_read_handle(_message_handle);
      self.time_increment = sensor_msgs_msg_laser_scan_time_increment_read_handle(_message_handle);
      self.scan_time = sensor_msgs_msg_laser_scan_scan_time_read_handle(_message_handle);
      self.range_min = sensor_msgs_msg_laser_scan_range_min_read_handle(_message_handle);
      self.range_max = sensor_msgs_msg_laser_scan_range_max_read_handle(_message_handle);
      }
    }
  }
}

impl rclrs_common::traits::Message for LaserScan {
  fn get_native_message(&self) -> uintptr_t {
    return self.get_native_message();
  }

  fn destroy_native_message(&self, message_handle: uintptr_t) -> () {
    self.destroy_native_message(message_handle);
  }

  fn read_handle(&mut self, message_handle: uintptr_t) -> () {
    self.read_handle(message_handle);
  }
}

impl rclrs_common::traits::MessageDefinition<LaserScan> for LaserScan {
  fn get_type_support() -> uintptr_t {
    return unsafe { sensor_msgs_msg_laser_scan_get_type_support() };
  }

  fn static_get_native_message(message: &LaserScan) -> uintptr_t {
    return message.get_native_message();
  }

  fn static_destroy_native_message(message_handle: uintptr_t) -> () {
    unsafe {
      sensor_msgs_msg_laser_scan_destroy_native_message(message_handle);
    }
  }
}
```

I'd love to help out, but this is currently beyond my Rust chops.

nnarain, 2020-09-26:
I think this where https://github.com/ros2-rust/ros2_rust/issues/21 comes in. That would give us fully support for for all the IDL message types introduced in Dashing.

nnmm, 2022-03-18:
Closed by #80.

closed 2022-03-18 by nnmm