ros2-rust/ros2_rust #28: Failed to colcon build with ros2:master source on Ubuntu20.04.

fujitatomoya, 2020-10-12:
### Colcon Build Failure

```
--- stderr: builtin_interfaces
CMake Error at /root/ros2_ws/ros2-rust_ws/install/rosidl_generator_rs/share/rosidl_generator_rs/cmake/rosidl_generator_rs_generate_interfaces.cmake:15 (find_package):
  By not providing "Findrmw_implementation_cmake.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rmw_implementation_cmake", but CMake did not find one.

  Could not find a package configuration file provided by
  "rmw_implementation_cmake" with any of the following names:

    rmw_implementation_cmakeConfig.cmake
    rmw_implementation_cmake-config.cmake

  Add the installation prefix of "rmw_implementation_cmake" to
  CMAKE_PREFIX_PATH or set "rmw_implementation_cmake_DIR" to a directory
  containing one of the above files.  If "rmw_implementation_cmake" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /root/ros2_ws/ros2-rust_ws/install/ament_cmake_core/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /root/ros2_ws/ros2-rust_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)

make: *** [Makefile:1392: cmake_check_build_system] Error 1
```

### Procedure

1. set up ros2 master colcon workspace with https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/
2. add https://github.com/ros2-rust/ros2_rust into colcon workspace
3. then `colcon build`

adamkrawczyk, 2021-03-13:
Have you found any way to overcome this issue?

fujitatomoya, 2021-03-14:
No not yet, i think there's been work on colcon-cargo before we can do this.

adamkrawczyk, 2021-03-15:
I got a slightly different compile error but thought that might be linked. 
Fortunately, I build for docker is passing on my fork (missed depend), but seems to not compile when another std_msgs lib is installed.

nnmm, 2022-02-19:
I'm closing this issue since the build system has drastically changed recently, so it is likely that this error does not occur anymore. Please reopen if you still encounter the error.

closed 2022-02-19 by nnmm