ros2-rust/ros2_rust #32: Should we use Corrosion for CMake?

jhdcs, 2020-11-05:
Rust has a crate called [corrosion](https://github.com/AndrewGaspar/corrosion#installation) that is supposed to simplify the build process when CMake is required. Would the CMake in our current project benefit from using this crate?

If we decide to use this, we will need to raise the minimum required CMake version to at least 3.12. However, using that version of CMake means that we won't need to include corrosion within the project - we could instead use CMake's FetchContent module (which came into being in version 3.11).

gbiggs, 2020-11-09:
I think it's better to work on fixing `colcon-cargo` so that we can build `rclrs` more "natively" within a colcon workspace. This is something I hope to work on over the next few weeks.

jhdcs, 2020-11-10:
Ah, perhaps I misunderstood, but I thought `colcon-cargo` was going to be used for importing non `rclrs` projects? If we are going to use `colcon-cargo` to build `rclrs`, though, I can stop working on the `corrosion` branch on my fork. I was getting stuck on getting the generated outputs into the `install` folder anyways...

nnarain, 2020-11-10:
I think the concept is that `colcon` (and extensions such as `colcon-cargo`) are not necessarily ROS2 build tools. `colcon-cargo`  doesn't appear to do anything specific to ROS2. It integrates `cargo` into `colcon`. For example `colcon-ros` appears to be the plugin that actually builds cmake ROS packages.

So if a ROS2 Rust project is build-able via `cargo`. `colcon-cargo` can be used to integrate that in a normal `colcon build` command.

If `corrosion` is used, `colcon-cargo` is not necessary since `colcon-ros` will build it as a `cmake` package.

gbiggs, 2020-11-10:
colcon is completely unspecific to ROS. It's an entirely separate project - it isn't even an OSRF project.

Like any other ROS client library, it's perfectly fine to build `rclrs` with `colcon-cargo`. Absolutely everything in ROS 2 is built using `colcon`, even colcon itself is built using `colcon`. That would eliminate an extra translation step that `corrosion` would introduce.

jhdcs, 2020-11-11:
Ah, excellent. Thank you both for the information.

I have additionally found that `corrosion` creates cargo libraries as interface libraries. Which is fine when you're linking that library to another existing library or executable, but exceedingly annoying if you're just trying to build it standalone (if you're interested, my efforts can be viewed [here](https://github.com/jhdcs/ros2_rust/tree/feature/corrosion_integration)). As such, in addition to your comments above, I will agree that `corrosion` is probably not the best fit here.

I will begin looking at `colcon-cargo`. Thank you both very much for your help!

esteve, 2020-11-11:
> Like any other ROS client library, it's perfectly fine to build rclrs with colcon-cargo. Absolutely everything in ROS 2 is built using colcon, even colcon itself is built using colcon. That would eliminate an extra translation step that corrosion would introduce.

@gbiggs @jhdcs That's not how it works. Take `rclpy` for example, it's built using CMake, not `setuptools`. The reason for that is that all the ament machinery is fully implemented in CMake (i.e. registering the `rcl` implementation, typesupport, exporting dependencies, etc.). However, packages that consume `rclrs` should be built via `colcon-cargo-ros` (which does not exist yet), which would expose the ROS extensions to `cargo` (see for reference https://github.com/colcon/colcon-ros/blob/master/colcon_ros/task/ament_python/build.py#L22 for a `colcon` task that builds `ament_python` packages).

jhdcs, 2020-11-11:
So in that case, I should continue working on getting `corrosion` to work? If so, I may require some help, since I'm getting a bit stuck getting the build outputs into the `install` folder...

esteve, 2020-11-12:
The `rclrs` CMake file does not compile anything, it just copies the source Rust files to a location in the ament index so that downstream packages can consume them (à la the crate index), so I don't know what `corrosion` brings to the table.

It's the downstream packages that compile the code, and those downstream packages are the ones that should be built with `colcon-cargo-ros`

jhdcs, 2020-11-12:
> The `rclrs` CMake file does not compile anything, it just copies the source Rust files to a location in the ament index so that downstream packages can consume them (à la the crate index), so I don't know what `corrosion` brings to the table.

Ah. I misunderstood the ReadMe, then. I thought it was saying that it was building the binaries, as in compiling them. If all we're doing is copying source files around, then `corrosion` will do nothing for us.

Is this how the other Client Libraries operate as well?

closed 2020-11-13 by jhdcs

esteve, 2020-11-18:
@jhdcs depends on the language of the client library. For example, the Python client library does not compile the Python source file, but it compiles the C bridge between ROS 2 and Python. Whereas the Java client library needs to compile the Java files to distribute them as JARs to the downstream packages. `rclrs` does not use an intermediate layer between ROS 2 and Rust as it uses FFI via `bindgen`.