ros2-rust/ros2_rust #34: Improve support for rust language analyzers

ruffsl, 2020-11-07:
During the last Rust WG meeting, members reported difficulties in trying to get rclrs to play nicely with VSCode + Rust-Analyzer. In particular, one of the hangups is that rust-analyzer gets confused by auto generated messages, and can’t recognize them as types. This could tie into some of the other tickets for improving message generation and build integration with cargo, although this ticket specifically tracks issues for improving support for rust language analyzers such as rls or rust-analyzer:

* https://github.com/rust-analyzer/rust-analyzer
* https://github.com/rust-lang/rls

This'll not only aid in rclrs development, but also help downstream users creating their own ROS Rust packages.

Related:
* https://github.com/ros2-rust/ros2_rust/issues/11
* https://github.com/ros2-rust/ros2_rust/issues/13

ruffsl, 2020-11-07:
@kjeremy @jhdcs @gbiggs , I recall you had started discussing this topic on the Rust WG proposal post along this thread:
https://discourse.ros.org/t/proposal-rust-working-group-for-ros/16993/30

kjeremy, 2020-11-07:
@ruffsl

As mentioned in that thread the issue is that the included examples copy the `Cargo.toml` and inject entries for the messages during build. If you open the build space folder in vscode then `rust-analyzer` will give you types. The easiest solution from the `rls`/`rust-analyzer` POV would be to include those in the `Cargo.toml` or maybe `build.rs` from the beginning.

jhdcs, 2020-11-10:
> The easiest solution from the `rls`/`rust-analyzer` POV would be to include those in the `Cargo.toml` or maybe `build.rs` from the beginning.

I think that this may be the best solution. I can forsee people using `rclrs` not knowing that they have to open the build space folder for `rclrs` in order to get types for whatever crate they're working on. This would be especially bad if they didn't download ROS 2 from source.

gbiggs, 2020-11-20:
This sounds like we should expect users to do with `Cargo.toml` what they have to do with `CMakeLists.txt` in a CMake or `ament_cmake` package: Include the message information in two places. It's as lovely a solution as it is in those other packages, but it seems it would solve the problem.

PR #64 (2021-11-22): Build system refactor
Closes #61, Closes #34, Closes #38, Closes #31, Closes #13

I have refactored the build system so that things can be build in a more Rust-like way: The biggest surprise people may have is that the `rclrs_examples` crate will generate a `.cargo/config.toml` file in order to patch in the file dependency for `std_msgs`.

Do the changes look acceptable? While I'm still not super-fond of the dependency search we have to do for `rclrs_examples`, I'm sure that we can find something else to make that easier in another PR.
closed 2021-11-22 by esteve