ros2-rust/ros2_rust #35: Missing Unit Tests

jhdcs, 2020-11-11:
Similar to #33, the code on a whole lacks unit testing. We should implement these tests sooner rather than later, since unit tests should ideally be created at around the same time the code that they're testing is made.

For the Rust portions of the project, I think the files are small enough that we can go with the standard Rustic way of adding unit testing in a module at the bottom of a file. Once the files get bigger, we can consider moving them off into their own folder. We will need to teach `colcon-cargo` how to run these tests, but that shouldn't be too difficult.

The other languages should follow the standard ROS 2 rules for unit testing.

ruffsl, 2021-01-19:
What kind of tests should we start with?
Are there any particular category of tests we could start with my mimicking/porting what exists for rclcpp or rclpy?

jhdcs, 2021-01-20:
I think that adding unit tests for `context`, as well as the node `Publisher` and `Subscriber` should probably be a priority. I'd put `context` at the top of the list, though - both because it is the lynchpin that the rest of the library builds off of (from what I'm able to see), and that it's what deals most directly with FFI with `rcl`.

Additionally, any function that has `unsafe` should have at _least_ one unit test. Ideally, if we can get away with it, I'd put more in there, just so we can make sure that we know that it fails in a way we expect.

A quick look for unsafe in functions turns up the following (though I may have missed a few):

1. `rclrs/src/context.rs` - `ContextHandle.drop()`
2. `rclrs/src/context.rs` - `Context.init()`
3. `rclrs/src/context.rs` - `Context.ok()`
4. `rclrs/src/context.rs` - `Context.default()`
5. `rclrs/src/lib.rs` - `spin()`
6. `rclrs/src/lib.rs` - `spin_once()`
7. `rclrs/src/node/mod.rs` - `NodeHandle.drop()`
8. `rclrs/src/node/mod.rs` - `Node.new_with_namespace()`
9. `rclrs/src/node/publisher.rs` - `PublisherHandle.drop()`
10. `rclrs/src/node/publisher.rs` - `Publisher.new()`
11. `rclrs/src/node/publisher.rs` - `Publisher.publish()`
12. `rclrs/src/node/subscription.rs` - `SubscriptionHandle.drop()`
13. `rclrs/src/node/subscription.rs` - `SubscriptionBase.take()`
14. `rclrs/src/node/subscription.rs` - `Subscription.new()`
15. `rclrs/src/node/subscription.rs` - `Subscription.take()`

PR #84 (2022-06-14): Add unit testing
Closes #35

Supersedes #52
closed 2022-06-14 by esteve