ros2-rust/ros2_rust #39: Use of second `ros2_rust` directory inside the `src` directory

bergercookie, 2020-11-24:
In the README, the following is provided for setting up the workspace:

```
mkdir -p ~/ros2_rust_ws/src/ros2-rust
cd ~/ros2_rust_ws/src/ros2-rust
git clone https://github.com/ros2-rust/ros2_rust.git
Why not clone it directly under src/?
```

Is there a reason for having nested ros2_rust directories, i.e., `src/ros2-rust/ros2_rust`?
Can we remove the intermediate to simplify the instructions?

ruffsl, 2021-01-19:
I like having the github repo name nested under the github org it comes from. It makes it a little more obvious on where repos are coming from if you start adding more packages into your workspace. E.g. repo files that clones into related org folders:

https://github.com/ros-planning/navigation2/blob/61710875ed1d73e807b7cd1a9e577d995f2e4ead/tools/underlay.repos

nnmm, 2022-02-19:
This has been changed as part of #73.

closed 2022-02-19 by nnmm