ros2-rust/ros2_rust #43: Make compatible with Rolling

jhdcs, 2021-09-03:
Since this client library is under such active development, perhaps we should make it so that it is compatible with / buildable on Rolling?

I will look into what needs to change for this to happen.

esteve, 2021-09-03:
> I will look into what needs to change for this to happen.

Thanks! That'd be great

jhdcs, 2021-09-08:
So, currently I'm running into an issue where I can't build `rclrs_examples`:
<details>
    <summary>Complete Colcon Build output</summary>
    
```
jhassold@952e670e3ad9:~/ros2_rust_ws$ colcon build
Starting >>> ament_cmake_export_crates
Finished <<< ament_cmake_export_crates [0.14s]                  
Starting >>> rclrs_common
Finished <<< rclrs_common [0.12s]                
Starting >>> rclrs
Starting >>> rosidl_generator_rs
Finished <<< rclrs [0.13s]                                                               
Finished <<< rosidl_generator_rs [0.52s]                  
Starting >>> rclrs_examples
--- stderr: rclrs_examples                              
 Downloading crates ...
  Downloaded syn v1.0.76
  Downloaded cc v1.0.70
warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:2392:10
     |
2392 |     ) -> u128;
     |          ^^^^ not FFI-safe
     |
     = note: `#[warn(improper_ctypes)]` on by default
     = note: 128-bit integers don't currently have a known stable ABI

warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:4101:18
     |
4101 |         __value: u128,
     |                  ^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:4109:18
     |
4109 |         __value: u128,
     |                  ^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:4117:18
     |
4117 |         __value: u128,
     |                  ^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:4144:18
     |
4144 |         __value: u128,
     |                  ^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/jhassold/ros2_rust_ws/build/rclrs_examples/ament_cargo/rclrs_examples/target/release/build/rclrs-d3b569f0be5f93c4/out/rcl_bindings.rs:4154:18
     |
4154 |         __value: u128,
     |                  ^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

warning: 6 warnings emitted

   Compiling rclrs_examples v0.1.0 (/home/jhassold/ros2_rust_ws/build/rclrs_examples)
error[E0432]: unresolved import `std_msgs`
 --> src/rclrs_publisher.rs:2:5
  |
2 | use std_msgs;
  |     ^^^^^^^^ no external crate `std_msgs`

error[E0432]: unresolved import `std_msgs`
 --> src/rclrs_subscriber.rs:2:5
  |
2 | use std_msgs;
  |     ^^^^^^^^ no external crate `std_msgs`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `rclrs_examples`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: build failed
make[2]: *** [CMakeFiles/build_crate.dir/build.make:63: ament_cargo/rclrs_examples/target/release/rclrs_publisher] Error 101
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/build_crate.dir/all] Error 2
make: *** [Makefile:129: all] Error 2
---
Failed   <<< rclrs_examples [0.74s, exited with code 2]

Summary: 4 packages finished [1.69s]
  1 package failed: rclrs_examples
  1 package had stderr output: rclrs_examples
  ```
  </details>
  
  So, apart from a bunch of FFI warnings (which can always be addressed later), it looks like somehow `colcon` isn't able to see the `rclrs` and `std_msgs` packages? Is that right? Or is something not getting passed to `cargo` correctly?
  
  I'm a little stumped at the moment, so any advice would be very helpful...

jhdcs, 2021-09-24:
Solved by #44

closed 2021-09-24 by jhdcs