ros2-rust/ros2_rust #45: Replace `failure` with `thiserror`

jhdcs, 2021-09-17:
The [`failure` crate](https://github.com/rust-lang-deprecated/failure) currently used by `ros2_rust` is deprecated. As such, we should probably move towards a different error-handling crate.

The `failure` crate maintainer notes in the readme that the [`thiserror` crate](https://github.com/dtolnay/thiserror) is a near drop-in replacement, so that may be the direction we want to take.

jhdcs, 2021-09-28:
With the merging of PR #46, this issue is resolved.

closed 2021-09-28 by jhdcs

esteve, 2021-09-28:
@jhdcs just a tip, when you open a PR, if it says "Closes #XXX", when the latter is merged, the issues will be closed automatically:

https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue