ros2-rust/ros2_rust #72: [Rolling] `rclrs` unable to build - cannot find headers for wrapper

jhdcs, 2022-02-01:
`rclrs` is failing to build on the latest version of Rolling - `bindgen` is unable to find the includes for `rcl` during compilation.

It is possible that this issue is related to https://github.com/ros2/rcl/pull/959. If so, the build script for `rclrs` will need to be tweaked to point to the new location of the headers.

sloretz, 2022-02-02:
> https://github.com/ros2-rust/ros2_rust/blob/dc454890e7cc26adb4dba1cc28703cfb00fb0b89/rclrs/build.rs#L32

Changing that to `-I{}/include/rcl` seems like it would update it to the new install directory.

Is there a way to tell bindgen to grab the include directory from CMake? Can it invoke a CMake script to grab the include directories off the `rcl::rcl` target?

jhdcs, 2022-02-02:
Yeah, I tried changing that to grab the inner install directory, but then it complains that `rcl/context.h` cannot find `rmw/init.h`... See https://github.com/jhdcs/ros2_rust/blob/bug/fix_missing_includes/rclrs/build.rs for the changed build script.

I am not sure how to pass information from CMake into the build script... That's where the information would get passed to `bindgen`. Specifically, the part of `bindgen` we're working with here is the Builder: https://docs.rs/bindgen/latest/bindgen/struct.Builder.html... I'll keep digging through that API to see what abilities we have to grab stuff from CMake, but I'm not sure there's all that much...

I've also looked for Rust crates to interact with CMake, but I was only able to find https://github.com/corrosion-rs/corrosion (which I wasn't _quite_  able to get working last time I poked at it), and https://crates.io/crates/cmake (which looks like bindings to allow Rust to call CMake - whereas we're trying to go the other direction here).

mxgrey, 2022-02-03:
Yeah, I've also been looking into this the last few days. Unfortunately everything I can find is indicating that cmake doesn't like to leak information outside of its own build system pipeline. There doesn't seem to be any way to just query cmake for the include directories of the target in a cmake package.

The most promising lead I could find for this was the [`cmake --find-package`](https://stackoverflow.com/a/28863712) command, but when I tested it out on a very simple package structure it did **not** provide any of the package target's transitive dependency headers, which is what we would need in this case.

As far as I can tell the `corrosion-rs` package that @jhdcs linked to is meant to do the inverse of what we want.. it allows Cargo Crates to be built as part of a cmake project, but it doesn't seem to provide a way to push information from cmake into Cargo.

esteve, 2022-02-03:
@jhdcs correct me if I'm wrong, but the problem is that we need to pass info from CMake to `cargo`, right?

In that case, we can set environment variables from CMake and when we invoke `cargo`, use `std::env::var` in the build script to read and add them to the include path for `clang`. We can obtain the include dirs for `rcl` with `$rcl_INCLUDE_DIRS` after calling `find_package`, or use `get_target_property` on the `rcl::rcl` target.

jhdcs, 2022-02-03:
@esteve That's part of the problem, yes. We get around it by digging through the `install` directory to find folders we need to include, but `CMake` doesn't really pass anything to us (apart from the `AMENT_PREFIX_PATH` environment variable, which is what we're currently using to find directories to include)

The weird thing is, even when we're including all these files (via `-I/path/to/install/directory/...`, it STILL doesn't build. Like, the included headers don't seem to be able to find each other...

mxgrey, 2022-02-03:
I was also considering feeding info to Rust from cmake using environment variables. My only hesitation with that is it locks `ros2_rust` into being a cmake package at its top level... indefinitely, and I don't find that palatable. I'm hoping to migrate away from `ros2_rust` needing cmake as a build tool as soon as feasible.

esteve, 2022-02-03:
@jhdcs what I meant is that we set the variables ourselves from CMake (e.g. `set(ENV{ROS2_RUST_RCL_INCLUDE_DIRS} path)`) and then call cargo, which will read them

esteve, 2022-02-03:
@mxgrey using CMake for `rclrs` is fine IMHO, it's what `rclpy` (https://github.com/ros2/rclpy/blob/master/rclpy/CMakeLists.txt) and the other bindings do, precisely because it's easier to query the ament info from CMake than from another language. However, programs that use `rclrs` should be buildable via `cargo` only (the same way that programs that use `rclpy` use standard Python build tools i.e. `setup.py`)

mxgrey, 2022-02-03:
Yeah, that's a fair outlook.

Even so, I don't know of a mechanism in cmake where you could say "Hey, let's pretend I have a target that depends on this imported target. Please tell me all the include flags that we would give the compiler." It would be great if that exists, but I haven't found it yet.

sloretz, 2022-02-03:
> I'm hoping to migrate away from ros2_rust needing cmake as a build tool as soon as feasible.

If `rcl` exported something like a pkg-config file, would using that be an acceptable solution?

>  "Hey, let's pretend I have a target that depends on this imported target. Please tell me all the include flags that we would give the compiler."

I don't know a good way, but if you want an ugly way:

```console
$ cat /home/sloretz/ws/ros2/install/rcl/share/rcl/cmake/rclExport.cmake | grep INTERFACE_INCLUDE_DIRECTORIES | awk -F '"' '{print $2}'
${_IMPORT_PREFIX}/include/rcl
```

mxgrey, 2022-02-04:
Yeah, having a pkg-config for `rcl` would be great. I only hesitate to ask for that because I know how much trouble it is after implementing it for `ign-cmake`. Without using a tool like `ign-cmake` the maintainers of `rcl` would need to make sure they update both the cmake information and the `pkg-config` information any time a dependency changes.

Maybe for `rcl` that's a worthwhile endeavor since it's meant to be accessible for bindings across many languages. And as long as the dependencies of `rcl` don't change very often, the maintenance shouldn't be a big deal.

mxgrey, 2022-02-04:
> if you want an ugly way:

This is roughly the same thing that doing `cmake --find-package ...` would provide. It doesn't include the transitive dependencies of imported targets, e.g. `rmw`, which is the troublesome part that we want cmake to provide.

PR #77 (2022-03-04): Bug/fix missing includes
Fixes #72 

Re-configured the build script to follow the new folder structure set in `rcl`. Also fixed a longstanding bug in the `allowlist_var` calls.
closed 2022-03-04 by jhdcs