ros2-rust/ros2_rust #93: spin() does not check which subscription has received messages

nnmm, 2022-04-02:
The spin() method simply calls `take()` on _all_ subscriptions, which fails when there is no new message. So this code only works for one subscription, not multiple.

nnmm, 2022-04-03:
I'll post a fix tomorrow.

esteve, 2022-04-03:
@nnmm this regression was introduced by #80, what was the rationale for removing the code that checks for incoming messages?

esteve, 2022-04-03:
For more context, `spin` must call `take` for all subscriptions, and only when there's data (i.e. `rcl_take` does not return an error), the message should be passed onto the callback registered for the subscription.

nnmm, 2022-04-03:
@esteve I think we should do this instead: https://github.com/ros2/rcl/blob/35a31b00a12f259d492bf53c0701003bd7f1745c/rcl/include/rcl/wait.h#L419

nnmm, 2022-04-03:
See here: https://github.com/ros2-rust/ros2_rust/pull/96

esteve, 2022-04-03:
> @esteve I think we should do this instead: https://github.com/ros2/rcl/blob/35a31b00a12f259d492bf53c0701003bd7f1745c/rcl/include/rcl/wait.h#L419

Also from the documentation there:

https://github.com/ros2/rcl/blob/35a31b00a12f259d492bf53c0701003bd7f1745c/rcl/include/rcl/wait.h#L387-L389=

```
 * For subscriptions this means there may be messages that can be taken, or
 * perhaps that the state of the subscriptions has changed, in which case
 * rcl_take may succeed but return with taken == false.
 ```

so, even if the subscription is ready, it may not contain a message. That's why the code before #80 checked if `rcl_take` returned `SubscriberErrorCode::SubscriptionTakeFailed`, in which case, it does not mean that `rcl_take` failed, but no message should was in the subscription and therefore the callback shouldn't be triggered, and that's also why `Subscription#take` returned a boolean instead of the message, to tell the caller that a message was taken.

nnmm, 2022-04-03:
Ah okay, then we should probably do both. I'll add it to my PR.

nnmm, 2022-04-03:
Done.

esteve, 2022-04-04:
Fixed via #96

closed 2022-04-04 by esteve