ros2-rust/ros2_rust #120: Use ndarray in messages

nnmm, 2022-04-22:
In Python, arrays of primitive numbers are deserialized into `numpy` arrays. Rust could do the same (in its idiomatic message type, not the RMW-native message type). This would make working with these vectors and matrices much easier.

There are two ways we could go about this: Either simply always use `ndarray` for arrays of numbers, or keep it a regular array and add conversion functions. The latter approach has the benefit that it could be feature-gated to avoid increasing compile times too much.

Maybe this whole idea could be taken a step further and we special-case the `MultiDimensionalArray` messages from `common_interfaces`, e.g. [Float64MultiArray](https://github.com/ros2/common_interfaces/blob/master/std_msgs/msg/Float64MultiArray.msg) to be `ndarray` types. I think `rclcpp` has a similar special-casing mechanism.

LaVieEstDure, 2022-07-26:
Is it not more idiomatic to just implement `Into` for MultiDimensionalArray messages of number types? The user can choose to `let val: Array3::<f64> = msg.into()` etc. if they desire.

nnmm, 2022-07-26:
Yep, that's probably better.