ros2-rust/ros2_rust #124: Create better examples

nnmm, 2022-04-22:
It would be nice to have demo applications that show something resembling useful work. They could showcase multi-threading, launch files, calling a Rust library from C++, and (once they've landed) services.

styczen, 2023-05-23:
I would like to work more on this issue. @nnmm could you provide more detailed information?

There are at least 3 repositories on [ROS2 Github organization](https://github.com/ros2):
- examples - https://github.com/ros2/examples,
- demos - https://github.com/ros2/examples,
- tutorials - https://github.com/ros2/tutorials (this one has not been worked on for a long time)

It might be a good idea to replicate some of those in Rust and more.

jhdcs, 2023-05-24:
While I'm not @nnmm, I would like to add in my two cents, if they would help.

All three should be implemented, eventually, however there are perhaps different priorities for them. While we already have a couple examples, we do not have any real demo applications. Perhaps we could try making a small demo application, and write a tutorial around it?

The danger with this is that the library is still in a bit of flux - major features are still coming out, and the "ergonomics" of the library are still fairly untested. It may be likely that the demos would need to be re-written, which may be less trivial to do than if it was just an additional example or two. However, at the same time, writing the demo application would be a good way of actually testing the features and ergonomics of the library.

I think that this is what the issue is intending to do. If @nnmm agrees, I can rename the issue to something like "Create demo applications", and make a new issue for creating tutorials. Also, anyone that would be working on this issue would be encouraged to create issues with findings about how to make the library easier to use - hopefully some of the spawned issues might be things that other contributors would be able to easily implement!

styczen, 2023-05-27:
@esteve @jhdcs For me renaming this issue and creating a new one with examples sounds good. I'll probably try writing demos with Turtlesim. Regarding the demo, probably the best option is to use simulation from [Turtlesim package repo](https://github.com/ros/ros_tutorials/tree/rolling/turtlesim) and implement teleop, mimic, draw_square. Later use some GUI library to port whole simulation because I have no experience with GUI programming in Rust, that's why I would like to try this after other parts are implemented.

rcywongaa, 2024-12-06:
How would you guys feel about using bevy as the GUI frontend for this hypothetical rust-turtlesim package?