ros2-rust/ros2_rust #132: Add support for constants in messages

nnmm, 2022-04-28:
Currently, constants (such as `string MY_CONSTANT_STRING="ros2_rust"`) are not supported yet by `rosidl_generator_rs`. It would be nice to support them, for feature parity with other languages.

Should they be `const` or `static` on the Rust side?

Nizerlak, 2022-07-27:
> Should they be `const` or `static` on the Rust side?

Is there any reason, why they should be `static` instead of `const`?

nnmm, 2022-07-27:
> > Should they be `const` or `static` on the Rust side?
> 
> Is there any reason, why they should be `static` instead of `const`?

Some of the constants have types that cannot be constructed as a const, e.g. string types, but with a static we could use the lazy initialization workaround (e. g. using https://crates.io/crates/once_cell).

nnmm, 2022-10-06:
Done in https://github.com/ros2-rust/ros2_rust/pull/269

closed 2022-10-06 by nnmm