ros2-rust/ros2_rust #139: Add support for node composition

nnmm, 2022-04-30:
See https://docs.ros.org/en/rolling/Concepts/About-Composition.html and https://docs.ros.org/en/rolling/Tutorials/Composition.html

esteve, 2022-05-02:
This one might be a bit tricky, but doable anyway. The tricky part is that in `rclcpp` and `rclpy`, nodes inherit from `Node`, which obviously we can't in Rust. The way I solved this in `rcljava` is by creating an interface called [ComposableNode](https://github.com/ros2-java/ros2_java/blob/main/rcljava/src/main/java/org/ros2/rcljava/node/ComposableNode.java), which contains a reference to the `Node` base class.

For Rust, we can have a similar trait, with a default implementation, e.g.

```rust

pub trait Node {
  fn create_publisher(&self, ...) {}
}

impl Node for MyNode {
  fn do_stuff(&self) {
    let publisher = self.create_publisher(...);
  }
}
```

nnmm, 2022-06-05:
This issue hinges on #126.

nnmm, 2022-08-20:
One thing I realized is that if we don't use a global context, component nodes will have to communicate across participants, which might be more expensive, not sure.