ros2-rust/ros2_rust #156: no matching package named `rclrs_example_msgs` found

maspe36, 2022-05-07:
Hi all,

I'm trying to build the project for the first time. Followed the instructions in the readme, but my build is currently failing.

```
sam@orange:~$ source /opt/ros/rolling/setup.bash 

sam@orange:~$ cd dev_ws/

sam@orange:~/dev_ws$ ll src
total 12
drwxrwxr-x  3 sam sam 4096 May  7 11:57 ./
drwxrwxr-x  7 sam sam 4096 May  7 12:10 ../
drwxrwxr-x 10 sam sam 4096 May  7 11:56 ros2_rust/

sam@orange:~/dev_ws$ colcon build 
Starting >>> rosidl_runtime_rs
Starting >>> rclrs_example_msgs
Finished <<< rosidl_runtime_rs [0.13s]                                
Starting >>> rclrs
Starting >>> rosidl_generator_rs
Finished <<< rclrs_example_msgs [0.29s]                                     
Finished <<< rosidl_generator_rs [0.24s]                  
Finished <<< rclrs [0.51s]                
Starting >>> rclrs_examples
--- stderr: rclrs_examples                         
error: no matching package named `rclrs_example_msgs` found
location searched: registry `crates-io`
required by package `rclrs_examples v0.2.0 (/home/sam/dev_ws/src/ros2_rust/rclrs_examples)`
---
Failed   <<< rclrs_examples [0.65s, exited with code 1]

Summary: 4 packages finished [1.42s]
  1 package failed: rclrs_examples
  1 package had stderr output: rclrs_examples

sam@orange:~/dev_ws$ pip show colcon-cargo
Name: colcon-cargo
Version: 0.1.2
Summary: Extension for colcon to support cargo packages.
Home-page: UNKNOWN
Author: Gérald Lelong
Author-email: gerald.lelong@easymov.fr
License: Apache License, Version 2.0
Location: /home/sam/.local/lib/python3.10/site-packages
Requires: catkin-pkg, colcon-cmake, colcon-core, colcon-library-path, colcon-pkg-config, colcon-python-setup-py, colcon-recursive-crawl, toml
Required-by: colcon-ros-cargo

sam@orange:~/dev_ws$ pip show colcon-ros-cargo
Name: colcon-ros-cargo
Version: 0.0.1
Summary: Extension for colcon to support Ament Cargo packages.
Home-page: https://colcon.readthedocs.io
Author: Nikolai Morin
Author-email: nnmmgit@gmail.com
License: Apache License, Version 2.0
Location: /home/sam/.local/lib/python3.10/site-packages
Requires: colcon-cargo, colcon-core, colcon-library-path, colcon-ros
Required-by: 

sam@orange:~/dev_ws$ colcon list
rclrs	src/ros2_rust/rclrs	(ament_cargo)
rclrs_example_msgs	src/ros2_rust/rclrs_example_msgs	(ros.ament_cmake)
rclrs_examples	src/ros2_rust/rclrs_examples	(ament_cargo)
rosidl_generator_rs	src/ros2_rust/rosidl_generator_rs	(ros.ament_cmake)
rosidl_runtime_rs	src/ros2_rust/rosidl_runtime_rs	(ament_cargo)

```

I tried source'ing `install/setup.bash` to get rclrs_example_msgs on my path, which works fine, but doesn't get the build any further. It seems like cargo is trying to find my rclrs_example_msgs package but since that's a regular ament_cmake package that seems wrong.

nnmm, 2022-05-07:
Thanks for the report. What's your OS, and what's the output of `colcon list`?

There are more detailed instructions on this branch, can you make sure you've followed them? https://github.com/ros2-rust/ros2_rust/blob/doc_articles/docs/Building.md

maspe36, 2022-05-07:
Unfortunately I actually missed a step in the readme..

`vcs import src < src/ros2_rust/ros2_rust_foxy.repos` and then source'ing just worked for me. 

Thanks for pointing me towards that branch, that looks very helpful!

closed 2022-05-07 by maspe36