ros2-rust/ros2_rust #161: Remove `create_node` from `Context`

esteve, 2022-05-13:
Neither the two official client libraries (`rclpy` and `rclcpp`) have a `create_node` method in their `Context` class. `rclcpy` has a `create_node` function at the module level, and `rclcpp` got rid of such function a while ago.

With both libraries when creating a node, they create a default context that gets assigned to the node if none is passed. With a builder pattern API we can achieve the same if needed. The goal of this ticket is to either move `create_node` back to the `rclrs` module or remove it altogether.

nnmm, 2022-05-13:
Are you proposing to use a global default context if none is given to the node builder?

nnmm, 2022-06-12:
@esteve Whether we also want a global default context is the crux of this proposal imo. I would like to keep the `create_node` method if users typically create a context, and a node based on that.