ros2-rust/ros2_rust #167: Warn or error on unknown ROS arguments

nnmm, 2022-05-16:
Currently, there is no warning for unknown ROS arguments, for instance:
```
cargo run --bin minimal_subscriber -- --ros-args --params-file=lol.yaml
```

This does _not_ load the parameter file lol.yaml, since ROS 2 doesn't support the `=` syntax here (it should be a space instead).

Nizerlak, 2022-07-03:
Depending on which approach would be chosen (warn or error) there should be logging (#184) implemented before or not. In case of warn prompting, logging mechanism is necessary, since error won't be returned from [Context::new()](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/context.rs#L62) and log call should happen there. Error raising variant would be simpler, since the only thing to do would be to add new enum to [RclrsError](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/error.rs#L8) (with Display implementation) and handle unparsed arguments similarly to non-ROS ones as in #147. 

I think that warn is more appropriate and logging mechanism should be implemented anyway, so in this case this issue will be blocked by #184. To make work more parallel maybe introduce bare logging interface without implementation, so call to it could be performed and this issue will be independent of #184 progress.

nnmm, 2022-07-04:
Yes, I agree logging is probably the way to go. Theoretically we could also make `Context::new()` return something like

```rust
struct ContextWithExtraInfo {
    context: Context,
    unparsed_ros_args: Vec<String>
}
```

but that's cumbersome to use.

Lucas-Rodrigues2, 2026-09-21:
Proposed approach for issue #167

Thanks for keeping this issue open. I would like to work on it.

The current code already has the internal `arguments::get_rcl_arguments` helper, and logging is now implemented. My proposed implementation is:

1. after `rcl_init` succeeds, query `rcl_context.global_arguments` with `rcl_arguments_get_count_unparsed_ros` and `rcl_arguments_get_unparsed_ros`;
2. after `LoggingLifecycle::configure` succeeds, emit one warning from the `rclrs.context` logger containing the unparsed ROS arguments in their original order;
3. keep non-ROS arguments unaffected;
4. add regression coverage for an unknown ROS argument such as `--params-file=missing.yaml`, plus a control case for a valid spelling using a separate argument.

I would prefer a warning rather than a new `RclrsError` variant because that matches the direction discussed above and avoids making context creation fail for arguments that `rcl` currently tolerates.

Before implementing, could you confirm:

- Is warning from `Context::new` still the desired behavior?
- Should the warning contain the complete ordered list in one message, or one message per argument?
- Would you prefer a public extraction API for unparsed ROS arguments, or should this remain internal to context initialization?

If this direction is accepted, I will submit a small test-first PR and run the repository formatting, clippy, and relevant test checks.

Assisted-by: Hermes Agent:gpt-5.6-sol