ros2-rust/ros2_rust #206: Add domain_id to node options

nnmm, 2022-06-23:
The `domain_id` is a useful option. In ROS distros up to Foxy, it is part of the node options. Afterwards it was moved to the context, or more precisely, the init options. See https://github.com/ros2-rust/ros2_rust/pull/186#discussion_r885689072

We should use the "ros_distro" cfg attribute to add it in the correct place for each distro.

The open question is, how do we pass this into the init options? Either we make `Context::new()` accept an `Option<RosDomainId>`, where `RosDomainId` is a `usize`, or we create a `ContextBuilder` the same way we created a `NodeBuilder`.

nnmm, 2022-06-23:
@Soya-Onishi You haven't yet started with this, correct? Would you be fine with it if I assign this issue to @DS3a?

Soya-Onishi, 2022-06-23:
@nnmm Sure! Please assign to @DS3a

esteve, 2025-08-21:
As of 0.5.0, we now have an API to set the domain id (https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/context.rs#L260-L263)

closed 2025-08-21 by esteve