ros2-rust/ros2_rust #207: Track all waitables that are currently in a wait set

nnmm, 2022-06-23:
During some refactoring to remove `SubscriptionBase`, I noticed that the wait set currently violates the unsafe code principles: Safe code can cause undefined behavior by adding the same item to multiple wait sets in different threads and waiting on them. From the `rcl` docs for `rcl_wait()`:

>  This function is thread-safe for unique wait sets with unique contents.
This function cannot operate on the same wait set in multiple threads, and the wait sets may not share content.
For example, calling rcl_wait() in two threads on two different wait sets that both contain a single, shared guard condition is undefined behavior.

We need to either make all functions that end up calling `rcl_wait()` unsafe, like `rclrs::spin()`, or we need to guarantee that wait sets do not share contents. The latter could be done by having a global list of waitables that are currently in a wait set, and would be my favored approach.

esteve, 2022-06-23:
@nnmm

> During some refactoring to remove `SubscriptionBase`

That is a major design decision, please file a ticket first so we can discuss this. A priori I'm not against, but I want to keep `Subscription` objects typed and I see how we can do that and remove `SubscriptionBase`.

nnmm, 2022-06-23:
@esteve I meant `SubscriptionHandle`, sorry. See https://github.com/ros2-rust/ros2_rust/pull/208

esteve, 2022-06-23:
@nnmm well, that PR also removes `SubscriptionBase` :stuck_out_tongue_winking_eye:

nnmm, 2022-06-25:
Another option is to do what `rclcpp` does and have an atomic bool `in_use_by_wait_set` inside each subscription/waitable.

PR #224 (2022-07-15): Ensure that a waitable is only added to one wait set at a time
Closes #207.

The use of atomic bools follows `rclcpp`'s approach.