ros2-rust/ros2_rust #213: Fix memory leaks

Nizerlak, 2022-07-03:
Recently I've run lib unit tests with valgrind for my [PR](https://github.com/ros2-rust/ros2_rust/pull/210) purposes and I've come across some memory leaks in rclrs lib. Here is full report:
```
==5444== 
==5444== HEAP SUMMARY:
==5444==     in use at exit: 43,376 bytes in 77 blocks
==5444==   total heap usage: 86,438 allocs, 86,361 frees, 22,290,504 bytes allocated
==5444== 
==5444== 64 bytes in 1 blocks are still reachable in loss record 1 of 20
==5444==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5444==    by 0x70B8C26: rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_typesupport_handle_function<rosidl_message_type_support_t>(rosidl_message_type_support_t const*, char const*) (in /opt/ros/foxy/lib/librosidl_typesupport_cpp.so)
==5444==    by 0x63DA8E1: ??? (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x63D8F4D: ??? (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x63D960F: ??? (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x63E37CB: rmw_create_node (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x48649C7: rcl_node_init (in /opt/ros/foxy/lib/librcl.so)
==5444==    by 0x141245: rclrs::node::builder::NodeBuilder::build (builder.rs:257)
==5444==    by 0x1665EB: rclrs::create_node (lib.rs:97)
==5444==    by 0x1586A9: rclrs::node::subscription::tests::test_instantiate_subscriber (subscription.rs:219)
==5444==    by 0x15EED0: rclrs::node::subscription::tests::test_instantiate_subscriber::{{closure}} (subscription.rs:216)
==5444==    by 0x13C40D: core::ops::function::FnOnce::call_once (function.rs:227)
==5444== 
==5444== 64 bytes in 1 blocks are still reachable in loss record 2 of 20
==5444==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5444==    by 0x4D19A61: rosidl_message_type_support_t const* rosidl_typesupport_c::get_typesupport_handle_function<rosidl_message_type_support_t>(rosidl_message_type_support_t const*, char const*) (in /opt/ros/foxy/lib/librosidl_typesupport_c.so)
==5444==    by 0x63EAE77: ??? (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x63E9DE1: rmw_create_subscription (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x486B3FE: rcl_subscription_init (in /opt/ros/foxy/lib/librcl.so)
==5444==    by 0x148800: rclrs::node::subscription::Subscription<T>::new (subscription.rs:106)
==5444==    by 0x16544A: rclrs::node::Node::create_subscription (node.rs:206)
==5444==    by 0x168360: rclrs::node::tests::test_create_subscription (node.rs:309)
==5444==    by 0x1335E0: rclrs::node::tests::test_create_subscription::{{closure}} (node.rs:305)
==5444==    by 0x13C4ED: core::ops::function::FnOnce::call_once (function.rs:227)
==5444==    by 0x1A8302: call_once<fn(), ()> (function.rs:227)
==5444==    by 0x1A8302: test::__rust_begin_short_backtrace (lib.rs:574)
==5444==    by 0x1A7068: call_once<(), (dyn core::ops::function::FnOnce<(), Output=()> + core::marker::Send), alloc::alloc::Global> (boxed.rs:1861)
==5444==    by 0x1A7068: call_once<(), alloc::boxed::Box<(dyn core::ops::function::FnOnce<(), Output=()> + core::marker::Send), alloc::alloc::Global>> (unwind_safe.rs:271)
==5444==    by 0x1A7068: do_call<core::panic::unwind_safe::AssertUnwindSafe<alloc::boxed::Box<(dyn core::ops::function::FnOnce<(), Output=()> + core::marker::Send), alloc::alloc::Global>>, ()> (panicking.rs:492)
==5444==    by 0x1A7068: try<(), core::panic::unwind_safe::AssertUnwindSafe<alloc::boxed::Box<(dyn core::ops::function::FnOnce<(), Output=()> + core::marker::Send), alloc::alloc::Global>>> (panicking.rs:456)
==5444==    by 0x1A7068: catch_unwind<core::panic::unwind_safe::AssertUnwindSafe<alloc::boxed::Box<(dyn core::ops::function::FnOnce<(), Output=()> + core::marker::Send), alloc::alloc::Global>>, ()> (panic.rs:137)
==5444==    by 0x1A7068: run_test_in_process (lib.rs:597)
==5444==    by 0x1A7068: test::run_test::run_test_inner::{{closure}} (lib.rs:491)
==5444== 
==5444== 64 bytes in 1 blocks are still reachable in loss record 3 of 20
==5444==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5444==    by 0x489F37C: rcutils_strndup (in /opt/ros/foxy/lib/librcutils.so)
==5444==    by 0x489E340: rcutils_load_shared_library (in /opt/ros/foxy/lib/librcutils.so)
==5444==    by 0x4D24D34: rcpputils::SharedLibrary::SharedLibrary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /opt/ros/foxy/lib/librcpputils.so)
==5444==    by 0x4D19A6F: rosidl_message_type_support_t const* rosidl_typesupport_c::get_typesupport_handle_function<rosidl_message_type_support_t>(rosidl_message_type_support_t const*, char const*) (in /opt/ros/foxy/lib/librosidl_typesupport_c.so)
==5444==    by 0x63EAE77: ??? (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x63E9DE1: rmw_create_subscription (in /opt/ros/foxy/lib/librmw_fastrtps_cpp.so)
==5444==    by 0x486B3FE: rcl_subscription_init (in /opt/ros/foxy/lib/librcl.so)
==5444==    by 0x148800: rclrs::node::subscription::Subscription<T>::new (subscription.rs:106)
==5444==    by 0x16544A: rclrs::node::Node::create_subscription (node.rs:206)
==5444==    by 0x168360: rclrs::node::tests::test_create_subscription (node.rs:309)
==5444==    by 0x1335E0: rclrs::node::tests::test_create_subscription::{{closure}} (node.rs:305)
==5444== 
==5444== 64 bytes in 1 blocks are still reachable in loss record 4 of 20
==5444==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==5444==    by 0x401677C: resize_scopes (dl-open.c:282)
==5444==    by 0x401677C: dl_open_worker (dl-open.c:693)
==5444==    by 0x4C1B927: _dl_catch_exception (dl-error-skeleton.c:208)
==5444==    by 0x4015609: _dl_open (dl-open.c:837)
==5444==    by 0x4AB634B: dlopen_doit (dlopen.c:66)
==5444==    by 0x4C1B927: _dl_catch_exception (dl-error-skeleton.c:208)
==5444==    by 0x4C1B9F2: _dl_catch_error (dl-error-skeleton.c:227)
==5444==    by 0x4AB6B58: _dlerror_run (dlerror.c:170)
==5444==    by 0x4AB63D9: dlopen@@GLIBC_2.2.5 (dlopen.c:87)
==5444==    by 0x489E35E: rcutils_load_shared_library (in /opt/ros/foxy/lib/librcutils.so)
==5444==    by 0x4D24D34: rcpputils::SharedLibrary::SharedLibrary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /opt/ros/foxy/lib/librcpputils.so)
==5444==    by 0x4D19A6F: rosidl_message_type_support_t const* rosidl_typesupport_c::get_typesupport_handle_function<rosidl_message_type_support_t>(rosidl_message_type_support_t const*, char const*) (in /opt/ros/foxy/lib/librosidl_typesupport_c.so)
[... 282 lines cut ...]
==5444==    by 0x19AF07: from_env (mod.rs:69)
==5444==    by 0x19AF07: new<std::io::stdio::Stdout> (mod.rs:159)
==5444==    by 0x19AF07: test::term::stdout (term.rs:33)
==5444==    by 0x187C23: test::console::run_tests_console (console.rs:266)
==5444==    by 0x19D226: test::test_main (lib.rs:115)
==5444==    by 0x19E200: test::test_main_static (lib.rs:134)
==5444==    by 0x166662: rclrs::main (lib.rs:1)
==5444==    by 0x13C44A: core::ops::function::FnOnce::call_once (function.rs:227)
==5444==    by 0x15A5AD: std::sys_common::backtrace::__rust_begin_short_backtrace (backtrace.rs:122)
==5444==    by 0x157EE0: std::rt::lang_start::{{closure}} (rt.rs:145)
==5444==    by 0x1E172D: call_once<(), (dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (function.rs:259)
==5444==    by 0x1E172D: do_call<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panicking.rs:492)
==5444==    by 0x1E172D: try<i32, &(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (panicking.rs:456)
==5444==    by 0x1E172D: catch_unwind<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panic.rs:137)
==5444==    by 0x1E172D: {closure#2} (rt.rs:128)
==5444==    by 0x1E172D: do_call<std::rt::lang_start_internal::{closure_env#2}, isize> (panicking.rs:492)
==5444==    by 0x1E172D: try<isize, std::rt::lang_start_internal::{closure_env#2}> (panicking.rs:456)
==5444==    by 0x1E172D: catch_unwind<std::rt::lang_start_internal::{closure_env#2}, isize> (panic.rs:137)
[... cut ...]
==5444== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 0 from 0)
```
Example places where leak occurs: [context.rs](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/context.rs#L80), [subscription.rs](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/node/subscription.rs#L106), [node.rs](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/node/builder.rs#L257).
Seems like deallocation is currently not utilized. For every `rcl_init` call there should be corresponding `deallocate` from `rcl_allocator`.

nnmm, 2022-07-04:
FYI @ros2-rust/dev 

Thanks for the report! I suspect most of these will be fixed by https://github.com/ros2-rust/ros2_rust/issues/188 – you probably terminated the application with `Ctrl-C`, right? Would be interesting to do this for a program that terminates voluntarily.

jhdcs, 2022-07-05:
> Most of them are related to loading the typesupport library instead.

Does this mean that there's potentially a leak in ROS 2? Or am I misreading this?

nnmm, 2022-07-07:
I ran the C++ talker/listener demo from Rolling in Valgrind and saw no memory leaks. But I'm also not seeing any memory leaks when I run the `rclrs` unit tests on the `main` branch:

```
$ valgrind --tool=memcheck --leak-check=full -- target/debug/deps/rclrs-bf2ac4d893bc907f
==699039== Memcheck, a memory error detector
==699039== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==699039== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==699039== Command: target/debug/deps/rclrs-bf2ac4d893bc907f
==699039==

running 11 tests
test context::tests::test_create_context ... ok
test context::tests::test_context_ok ... ok
test parameter::value::tests::test_parameter_value ... ok
test parameter::override_map::tests::test_resolve_parameter_overrides ... ok
test node::subscription::tests::test_instantiate_subscriber ... ok
test node::publisher::tests::test_publish_message ... ok
test node::publisher::tests::test_new_publisher ... ok
test tests::test_spin_once ... ok
test node::tests::test_create_publisher ... ok
test node::tests::test_create_subscription ... ok
test node::tests::test_new ... ok

test result: ok. 11 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 15.30s

==699039==
==699039== HEAP SUMMARY:
==699039==     in use at exit: 32,660 bytes in 86 blocks
==699039==   total heap usage: 42,439 allocs, 42,353 frees, 16,481,316 bytes allocated
==699039==
==699039== LEAK SUMMARY:
==699039==    definitely lost: 0 bytes in 0 blocks
==699039==    indirectly lost: 0 bytes in 0 blocks
==699039==      possibly lost: 0 bytes in 0 blocks
==699039==    still reachable: 32,660 bytes in 86 blocks
==699039==         suppressed: 0 bytes in 0 blocks
==699039== Reachable blocks (those to which a pointer was found) are not shown.
==699039== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==699039==
==699039== For lists of detected and suppressed errors, rerun with: -s
==699039== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

Which ROS distribution are you using, @Nizerlak?

Nizerlak, 2022-07-07:
I'm using Foxy.

ThomasHauth, 2023-02-18:
Hey, I ran valgrind on the unit tests with ros2-foxy on the latest main commit 087527ef12a3 with the latest Valgrind-3.21.0.GIT and there are no losses reported. Therefore, I recommend to close this ticket.
If you think its useful, I can make a PR to add a valgrind check to the CI of this repo.

```
# valgrind --tool=memcheck --leak-check=full -- target/debug/deps/rclrs-3249c5f110715714
==23853== Memcheck, a memory error detector
==23853== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==23853== Using Valgrind-3.21.0.GIT and LibVEX; rerun with -h for copyright info
==23853== Command: target/debug/deps/rclrs-3249c5f110715714
==23853== 

running 18 tests
test context::tests::test_create_context ... ok
test context::tests::context_is_send_and_sync ... ok
test context::tests::test_context_ok ... ok
test arguments::tests::test_non_ros_arguments ... ok
test arguments::tests::test_empty_non_ros_arguments ... ok
test node::tests::node_is_send_and_sync ... ok
test parameter::value::tests::test_parameter_value ... ok
test subscription::message_info::tests::wait_set_is_send_and_sync ... ok
test wait::exclusivity_guard::tests::test_exclusivity_guard ... ok
test wait::guard_condition::tests::test_guard_condition ... ok
test parameter::override_map::tests::test_resolve_parameter_overrides ... ok
test wait::guard_condition::tests::test_guard_condition_is_send_and_sync ... ok
test wait::guard_condition::tests::test_guard_condition_wait ... ok
test wait::tests::guard_condition_in_wait_set_readies ... ok
test wait::tests::wait_set_is_send_and_sync ... ok
test node::graph::tests::test_node_names_with_enclaves ... ok
test node::graph::tests::test_graph_empty ... ok
test node::graph::tests::test_node_names ... ok

test result: ok. 18 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 7.15s

==23853== 
==23853== HEAP SUMMARY:
==23853==     in use at exit: 39,812 bytes in 62 blocks
==23853==   total heap usage: 12,089 allocs, 12,027 frees, 8,154,987 bytes allocated
==23853== 
==23853== LEAK SUMMARY:
==23853==    definitely lost: 0 bytes in 0 blocks
==23853==    indirectly lost: 0 bytes in 0 blocks
==23853==      possibly lost: 0 bytes in 0 blocks
==23853==    still reachable: 39,812 bytes in 62 blocks
==23853==         suppressed: 0 bytes in 0 blocks
==23853== Reachable blocks (those to which a pointer was found) are not shown.
==23853== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==23853== 
==23853== For lists of detected and suppressed errors, rerun with: -s
==23853== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

```

Interesting find is that Valgrind-3.15.0 available in the `ros:foxy` Docker container reports an deviation is a syscall done by rust:

```
# valgrind --tool=memcheck --leak-check=full  --show-leak-kinds=all -- target/debug/deps/rclrs-3249c5f110715714
==134965== Memcheck, a memory error detector
==134965== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==134965== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==134965== Command: target/debug/deps/rclrs-3249c5f110715714
==134965== 
==134965== Syscall param statx(file_name) points to unaddressable byte(s)
==134965==    at 0x4B6988E: statx (statx.c:29)
==134965==    by 0x205E81: statx (weak.rs:176)
==134965==    by 0x205E81: std::sys::unix::fs::try_statx (fs.rs:159)
==134965==    by 0x2025A9: std::sys::unix::fs::stat (fs.rs:1298)
==134965==    by 0x1BA50B: metadata<&std::path::PathBuf> (fs.rs:1649)
==134965==    by 0x1BA50B: get_dbpath_for_term (searcher.rs:50)
==134965==    by 0x1BA50B: from_name (mod.rs:83)
==134965==    by 0x1BA50B: from_env (mod.rs:69)
==134965==    by 0x1BA50B: new<std::io::stdio::Stdout> (mod.rs:159)
==134965==    by 0x1BA50B: test::term::stdout (term.rs:33)
==134965==    by 0x1A74A3: test::console::run_tests_console (console.rs:258)
==134965==    by 0x1BC816: test::test_main (lib.rs:113)
==134965==    by 0x1BD7C0: test::test_main_static (lib.rs:132)
==134965==    by 0x18DC92: rclrs::main (lib.rs:1)
==134965==    by 0x1816DA: core::ops::function::FnOnce::call_once (function.rs:248)
==134965==    by 0x17228D: std::sys_common::backtrace::__rust_begin_short_backtrace (backtrace.rs:122)
==134965==    by 0x159A10: std::rt::lang_start::{{closure}} (rt.rs:145)
==134965==    by 0x1F56DD: call_once<(), (dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (function.rs:280)
==134965==    by 0x1F56DD: do_call<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panicking.rs:492)
==134965==    by 0x1F56DD: try<i32, &(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe)> (panicking.rs:456)
==134965==    by 0x1F56DD: catch_unwind<&(dyn core::ops::function::Fn<(), Output=i32> + core::marker::Sync + core::panic::unwind_safe::RefUnwindSafe), i32> (panic.rs:137)
==134965==    by 0x1F56DD: {closure#2} (rt.rs:128)
==134965==    by 0x1F56DD: do_call<std::rt::lang_start_internal::{closure_env#2}, isize> (panicking.rs:492)
==134965==    by 0x1F56DD: try<isize, std::rt::lang_start_internal::{closure_env#2}> (panicking.rs:456)
==134965==    by 0x1F56DD: catch_unwind<std::rt::lang_start_internal::{closure_env#2}, isize> (panic.rs:137)
==134965==    by 0x1F56DD: std::rt::lang_start_internal (rt.rs:128)
==134965==  Address 0x0 is not stack'd, malloc'd or (recently) free'd

```

However, this usage of the statx is not reported any more as an error in newer Valgrind versions. See [here](https://github.com/rust-lang/rust/issues/68979) and [here](https://sourceware.org/git/?p=valgrind.git;a=commit;h=2a7d3ae768f9e5b29acd5cb743c3fb13640a391c)

ThomasHauth, 2023-02-20:
@nnmm @Nizerlak can we close this issue? Thanks!

nnmm, 2023-02-21:
@ThomasHauth Thanks for following up on this issue! Yes, we can close it then.

closed 2023-02-21 by nnmm