ros2-rust/ros2_rust #215: Use std::os::raw types instead of libc crate

nnmm, 2022-07-04:
It seems that the libc crate is only really useful in `#[no_std]` environments. https://stackoverflow.com/questions/44436515/should-i-use-libcc-char-or-stdosrawc-char

We should see if we can replace all uses of `libc` with `std::os::raw`.

Tacha-S, 2022-10-15:
I found libc crate use of the following items within this project.

- `c_void`
  - It is used in `*mut c_void`, in which case both [libc](https://docs.rs/libc/latest/libc/enum.c_void.html) and [std::os::raw](https://doc.rust-lang.org/nightly/std/os/raw/type.c_void.html) are equivalent to C's `void*`.
- `c_char`(Both libc and std are used.)
  - Exist in [std::os::raw](https://doc.rust-lang.org/nightly/std/os/raw/type.c_char.html).
- `c_ushort`
  - Exist in [std::os::raw](https://doc.rust-lang.org/nightly/std/os/raw/type.c_ushort.html).
- `uintptr_t`
  - Not implemented in `std::os::raw`, but may be [replaced by `usize`](https://rust-lang.github.io/unsafe-code-guidelines/layout/scalars.html#isize-and-usize).
- `realloc`
  - Not implemented in `std::os::raw`. Compatibility with C is not specified, but [std::alloc::realloc](https://doc.rust-lang.org/std/alloc/fn.realloc.html) is the implementation in Rust.
- `memcpy`
  - Not implemented in `std::os::raw`, but [`copy_nonoverlapping`](https://doc.rust-lang.org/std/ptr/fn.copy_nonoverlapping.html) is semantically equivalent to C `memcopy`.
- `size_t`
  - Not implemented in `std::os::raw`. However, since `--size_t_is_usize` is enabled in bindgen, it can be replaced by `usize`.([ref](https://internals.rust-lang.org/t/pre-rfc-usize-is-not-size-t/15369))

https://github.com/ros2-rust/ros2_rust/blob/0c262973b5abe0b8b66bca0285fae3ec59c0176f/rclrs/build.rs#L37

I consider `c_void`, `c_char`, `c_ushort`, `uintptr_t` to be fine, what about `realloc`, `memcpy`, `size_t`?

nnmm, 2022-10-15:
I agree about replacing the integer aliases like you say. We could also use `std::ffi` instead of `std::os::raw`, I actually think that's a bit nicer.

For the allocation functions, they are currently only used in one place iirc: In `sequence.rs`, as a workaround for some functions that were missing in older distros. I had already backported them in https://github.com/ros2/rosidl/pull/667 but they hadn't landed in a Foxy release yet when I wrote `sequence.rs`. However, I just checked and they've now been released, so there is no need for the workaround anymore, and therefore for `realloc` and `memcpy`.

Tacha-S, 2022-10-16:
Thank you for the background on `realloc` and `memcpy`.

I will create a draft with the replacement with `std::ffi` and the elimination of `realloc` and `memcpy`.

nnmm, 2022-10-19:
Closed in #284

closed 2022-10-19 by nnmm