ros2-rust/ros2_rust #216: Replace parking_lot mutexes with stdlib

nnmm, 2022-07-04:
Starting with Rust 1.62, the standard lib mutexes have roughly equal performance to parking_lot: https://github.com/rust-lang/rust/pull/95035

IMO we could replace parking_lot with the standard lib mutexes throughout our code.

jhdcs, 2022-07-05:
This looks pretty good to me! It will help reduce the size of the library as well!

The only snag I can foresee is that we will have to deal with the possibility of `poison`ed Mutexes again, unless `std` did away with that?

nnmm, 2022-07-05:
Yes, the `lock()` calls will be slightly uglier, but lock poisoning is a safety feature, so I'd say it's a net win.

esteve, 2022-07-05:
Sounds good to me. The more we can use from `stdlib` the better :slightly_smiling_face:

PR #232 (2022-07-24): Replace parking_lot Mutex with std::sync::Mutex
This closes #216
closed 2022-07-24 by nnmm