ros2-rust/ros2_rust #225: colcon build fails with error "failed to resolve patches for `https://github.com/rust-lang/crates.io-index`"

VRichardJP, 2022-07-13:
I am folloowing the build documentation using the foxy docker image (https://github.com/ros2-rust/ros2_rust/blob/main/docs/building.md#building-with-colcon)

when I run  `colcon build` to build all the default packages, I get the following error:
```
root@6f6e313c6d9a:/workspace# colcon build
Starting >>> rosidl_runtime_rs
Starting >>> test_interface_files
Finished <<< rosidl_runtime_rs [0.09s]
Starting >>> rosidl_generator_rs
Finished <<< test_interface_files [0.10s]                                                            
Finished <<< rosidl_generator_rs [0.15s]                   
Starting >>> rosidl_default_runtime
Starting >>> rosidl_default_generators
Finished <<< rosidl_default_runtime [0.09s]                                   
Finished <<< rosidl_default_generators [0.09s]
Starting >>> builtin_interfaces
Starting >>> unique_identifier_msgs
Starting >>> rclrs_example_msgs
Starting >>> std_srvs
Starting >>> lifecycle_msgs
Finished <<< unique_identifier_msgs [0.48s]                                                                   
Finished <<< builtin_interfaces [0.52s]
Starting >>> std_msgs
Starting >>> action_msgs
Starting >>> rcl_interfaces
Starting >>> rosgraph_msgs
Starting >>> statistics_msgs
Finished <<< std_srvs [0.55s]                                                                          
Finished <<< rclrs_example_msgs [0.77s]                                                                      
Finished <<< lifecycle_msgs [0.98s]                                                               
Finished <<< rosgraph_msgs [0.53s]                                                                
Finished <<< action_msgs [0.72s]                                                                
Starting >>> example_interfaces
Starting >>> test_msgs
Finished <<< statistics_msgs [0.73s]                                                                 
Finished <<< rcl_interfaces [1.52s]                                                                    
Starting >>> composition_interfaces
Finished <<< std_msgs [1.66s]                                                                          
Starting >>> geometry_msgs
Starting >>> rclrs
Starting >>> actionlib_msgs
--- stderr: rclrs                                                                                          
error: failed to resolve patches for `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to load source for dependency `builtin_interfaces`

Caused by:
  Unable to update /workspace/install/builtin_interfaces/share/builtin_interfaces/rust

Caused by:
  failed to parse manifest at `/workspace/install/builtin_interfaces/share/builtin_interfaces/rust/Cargo.toml`

Caused by:
  namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
---
Failed   <<< rclrs [0.12s, exited with code 1]
Aborted  <<< composition_interfaces [0.44s]                                              
Aborted  <<< example_interfaces [1.30s]
Aborted  <<< actionlib_msgs [0.46s]                                                                             
Aborted  <<< test_msgs [1.56s]                                                                              
Aborted  <<< geometry_msgs [0.86s]                                 

Summary: 15 packages finished [3.62s]
  1 package failed: rclrs
  5 packages aborted: actionlib_msgs composition_interfaces example_interfaces geometry_msgs test_msgs
  1 package had stderr output: rclrs
  12 packages not processed

```

VRichardJP, 2022-07-13:
```
root@01bf40ebf5fa:/workspace# cat /workspace/install/builtin_interfaces/share/builtin_interfaces/rust/Cargo.toml
[package]
name = "builtin_interfaces"
version = "1.0.0"
edition = "2021"

[dependencies]
libc = "0.2"
rosidl_runtime_rs = "*"
serde = { version = "1", optional = true, features = ["derive"] }

[features]
serde = ['dep:serde', 'rosidl_runtime_rs/serde']
```

VRichardJP, 2022-07-14:
I modified the Dockerfile and removed the `--default-toolchain 1.59.0` in order to get the latest toolchain. Seems to have fixed my issue.
Should the dockerfile be updated? (any reason to force specific version number?)

nnmm, 2022-07-14:
Yes, we should either update the Dockerfile or rewrite https://github.com/ros2-rust/ros2_rust/blob/a635b0b7666d0676e671ee8768580e5477cdd664/rosidl_generator_rs/resource/Cargo.toml.em#L16 to not use the relatively new `dep:` syntax. If you want, feel free to create an MR.