ros2-rust/ros2_rust #226: Unit test segfault

nnmm, 2022-07-14:
To reproduce, run `cargo test`, note down the path of the test binary and run it several times, e.g. 

until target/debug/deps/rclrs-786fb1931d2745ba; [ $? -eq 139 ]; do printf '.'; done

nnmm, 2022-07-14:
Backtrace: 
```
#0  0x00007ff7b5e56b86 in eprosima::fastrtps::types::TypeIdentifier::operator=(eprosima::fastrtps::types::TypeIdentifier const&) () from /opt/ros/rolling/lib/libfastrtps.so.2.6
#1  0x00007ff7b5e8569b in ?? () from /opt/ros/rolling/lib/libfastrtps.so.2.6
#2  0x00007ff7b5e7f377 in register_builtin_annotations_types(eprosima::fastrtps::types::TypeObjectFactory*) () from /opt/ros/rolling/lib/libfastrtps.so.2.6
#3  0x00007ff7b5e74e63 in eprosima::fastrtps::types::TypeObjectFactory::get_instance() () from /opt/ros/rolling/lib/libfastrtps.so.2.6
#4  0x00007ff7b62c3e35 in rmw_fastrtps_shared_cpp::register_type_object(rosidl_message_type_support_t const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/rolling/lib/librmw_fastrtps_shared_cpp.so
#5  0x00007ff7b631c88a in ?? () from /opt/ros/rolling/lib/librmw_fastrtps_cpp.so
#6  0x00007ff7b6309777 in ?? () from /opt/ros/rolling/lib/librmw_fastrtps_cpp.so
#7  0x00007ff7b6310800 in rmw_create_node () from /opt/ros/rolling/lib/librmw_fastrtps_cpp.so
#8  0x00007ff7b6e6687a in rcl_node_init () from /opt/ros/rolling/lib/librcl.so
#9  0x0000564a6e00eca5 in rclrs::node::builder::NodeBuilder::build (self=0x7ff7b4b70ad8) at src/node/builder.rs:257
#10 0x0000564a6e03da9c in rclrs::node::Node::new (context=0x7ff7b4b70b78, node_name="Bob") at src/node.rs:103
```

So, probably FastRTPS' fault.

nnmm, 2022-07-14:
@ros2-rust/dev We should see if we can find an upstream ticket for this and then close this.

nnmm, 2022-07-14:
Potentially related to https://github.com/eProsima/Fast-DDS/issues/2635

maspe36, 2024-10-14:
This was fixed with https://github.com/ros2-rust/ros2_rust/pull/386

TL;DR we added additional protection around `rcl` calls because it uses globals internally

closed 2024-10-14 by maspe36