ros2-rust/ros2_rust #244: Add support for actions

nnmm, 2022-07-31:
Actions are essentially pub/sub + clients/services, both of which are already implemented, so this should be relatively straightforward.

esteve, 2023-02-26:
@Tacha-S I hadn't received a notification for your comment, I'm terribly sorry for the slow response. Are you still interested in actions? I started the work in https://github.com/ros2-rust/ros2_rust/pull/295, which has a generator for action structures, but it's still missing the code for actually sending/processing actions. Would you be interested in contributing to it? Thanks.

Tacha-S, 2023-04-09:
I don't have much free time these days either, but I haven't lost interest, and I will submit a PR if I can make it to the point where I can contribute regarding sending/processing actions.