ros2-rust/ros2_rust #245: Create a point cloud iterator package

nnmm, 2022-08-02:
There should be something like https://gitlab.com/ApexAI/point_cloud_msg_wrapper in the Rust ecosystem – i.e. a convenient and as-safe-as-possible way to interact with point clouds that doesn't incur a copy.

This requires good knowledge of `unsafe` and transmutation.

nnmm, 2022-08-02:
fyi @ruffsl

stelzo, 2023-05-01:
@nnmm I created something like this last week for using it with rosrust but it is a general solution for `sensor_msgs/PointCloud2` and rosrust_msg is just an optional feature. There is also no `unsafe` at all. It should work with other packages as long as the byte buffer `data` can be easily converted to `Vec` without a copy. Here it is: https://github.com/stelzo/ros_pointcloud2.

Hopefully this helps - I started learning Rust 2 months ago, so the code is probably not up to some standards but I am willing to learn!