ros2-rust/ros2_rust #246: Add Extend and FromIterator impl for rosidl_runtime_rs::{String, WString}

nnmm, 2022-08-14:
It would be nice if the string types in `rosidl_runtime_rs` could be built/extended from iterators, [like the regular string type](https://doc.rust-lang.org/std/string/struct.String.html#impl-Extend%3Cchar%3E).

For this, we'd need implementations for `Extend<char>` and `FromIterator<char>` for both types, at least.

maspe36, 2022-11-11:
An important thing to note, is that this feature gives the string types interior mutability. So these string types would have to lose the Sync impl.

https://github.com/ros2-rust/ros2_rust/blob/main/rosidl_runtime_rs/src/string.rs#L230

nnmm, 2022-11-11:
I think we wouldn't need to lose the `Sync` marker, since the `Extend` trait takes `&mut self` and not `&self`.

maspe36, 2022-11-12:
Ahh, gotcha. Yeah even rust's `String` is `Sync` 
https://doc.rust-lang.org/std/string/struct.String.html#impl-Sync-for-String

Updated in the PR

maspe36, 2023-03-04:
I believe this issue should be closed now that #293 has been merged

closed 2023-03-07 by nnmm