ros2-rust/ros2_rust #253: Issue of Republisher tutorial

soblin, 2022-09-11:
I'm referring to `republisher` tutorial [in this page](https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md) as an example of rclrs node with publisher and subscriber. But when I tried to compile the final code, it gives following error (happened on Ubuntu20, with rustc 1.60.0)

```
error[E0277]: `*const rclrs::rcl_bindings::rosidl_message_type_support_t` cannot be shared between threads safely
   --> src/main.rs:47:5
    |
47  |     std::thread::spawn(move || -> Result<(), rclrs::RclrsError> {
    |     ^^^^^^^^^^^^^^^^^^ `*const rclrs::rcl_bindings::rosidl_message_type_support_t` cannot be shared between threads safely
    |
    = help: within `RepublisherNode`, the trait `Sync` is not implemented for `*const rclrs::rcl_bindings::rosidl_message_type_support_t`
    = note: required because it appears within the type `Publisher<std_msgs::msg::String>`
note: required because it appears within the type `RepublisherNode`
   --> src/main.rs:4:8
    |
4   | struct RepublisherNode {
    |        ^^^^^^^^^^^^^^^
    = note: required because of the requirements on the impl of `Send` for `Arc<RepublisherNode>`
    = note: required because it appears within the type `[closure@src/main.rs:47:24: 53:6]`
note: required by a bound in `spawn`

error[E0277]: `*const rclrs::rcl_bindings::rosidl_message_type_support_t` cannot be sent between threads safely
   --> src/main.rs:47:5
    |
47  |     std::thread::spawn(move || -> Result<(), rclrs::RclrsError> {
    |     ^^^^^^^^^^^^^^^^^^ `*const rclrs::rcl_bindings::rosidl_message_type_support_t` cannot be sent between threads safely
    |
    = help: within `RepublisherNode`, the trait `Send` is not implemented for `*const rclrs::rcl_bindings::rosidl_message_type_support_t`
    = note: required because it appears within the type `Publisher<std_msgs::msg::String>`
note: required because it appears within the type `RepublisherNode`
   --> src/main.rs:4:8
    |
4   | struct RepublisherNode {
    |        ^^^^^^^^^^^^^^^
    = note: required because of the requirements on the impl of `Send` for `Arc<RepublisherNode>`
    = note: required because it appears within the type `[closure@src/main.rs:47:24: 53:6]`
note: required by a bound in `spawn`
```

Can you sincerely educate me on how to work with this error ? (Or is it expected result ?)

jhdcs, 2022-09-12:
This sounds like an error that should be fixed in #254, which should hopefully be merged in the near future.

nnmm, 2022-09-12:
Oh, I hadn't even seen this issue before making #254. But yes, it should be fixed now, feel free to reopen if it's not. And thanks for the report!

closed 2022-09-12 by nnmm