ros2-rust/ros2_rust #256: `std_msgs` not found

whiteplatin, 2022-09-12:
```
cargo check
    Updating crates.io index
error: no matching package named `std_msgs` found
location searched: registry `crates-io`
required by package `republisher_node v0.1.0 
```
https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md#create-a-package

AlexKaravaev, 2022-09-12:
Probably you haven't followed instructions in [README](https://github.com/ros2-rust/ros2_rust). common_interfaces must be inside workspace and be built.

whiteplatin, 2022-09-12:
where and how is that mentioned in the [README](https://github.com/ros2-rust/ros2_rust)

esteve, 2022-09-12:
@whiteplatin right [here](https://github.com/ros2-rust/ros2_rust#sounds-great-how-can-i-try-this-out):

```vcs import src < src/ros2_rust/ros2_rust_foxy.repos```

Please install ROS 2 completely and follow the instructions on the README.md, from your other ticket #255 it looks like you don't have all the dependencies of ROS 2 installed.

esteve, 2022-09-12:
> I only followed `https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html`, there was no mention of colcon apart from another page.

That page describes the setup for a ROS 2 end user, if you want to develop applications that use ROS 2, you'd need to follow https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries.html

> Well regarding #255 , I already had `python3-colcon-package-information` (and python3-colcon-common-extensions) before I opened that ticket.

That's rather strange, because the list verb is definitely in `python3-colcon-package-information`:

```sh
$ dpkg -S /usr/lib/python3/dist-packages/colcon_package_information/verb/list.py
python3-colcon-package-information: /usr/lib/python3/dist-packages/colcon_package_information/verb/list.py
```

In any case, `colcon` is not part of `ros2-rust`, so you should file a ticket in their repository if your issue persists.

> Is the issue humble ?

Doesn't seem so, `common_interfaces` is part of https://github.com/ros2-rust/ros2_rust/blob/main/ros2_rust_humble.repos#L2 and our CI is running for Foxy, Galactic, Humble and Galactic https://github.com/ros2-rust/ros2_rust/actions/runs/2964305847

closed 2022-09-12 by whiteplatin