ros2-rust/ros2_rust #263: How to use c++ headers of other ros packages ?

whiteplatin, 2022-09-20:
Hello,
is there a guide or has anyone used MoveIt2 with ros2_rust ?

Example:
https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html
```
#include <memory>

#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "hello_moveit",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );

  // Create a ROS logger
  auto const logger = rclcpp::get_logger("hello_moveit");

  // Next step goes here

  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}
```

Any ideas on how to get started ?
Do I need to use cxx for example or can I take different route to use this with ros2_rust ?

nnmm, 2022-09-20:
cc @tylerjw

In general, I don't think we're there yet. The two major options are (1) writing C wrappers for using either of C++/Rust in the other language or (2) trying out `cxx`.

What would you like to do more concretely?

tylerjw, 2022-09-20:
Hello, this is something I really want and have been trying to make possible but right now it is relatively involved as @nnmm stated. I created this: https://github.com/tylerjw/add_cargo_library

which can make it easier to add cxx rust code into your C++ code. This does not automate creating bindings, you'll have to create that boundary layer yourself with cxx.

This also does not make MoveIt work with ros2-rust as ros2-rust is a rust client library and MoveIt depends on the C++ client library. There is no bridging of the types between these two libraries so you can't for example create a node in ros2-rust and pass it to MoveIt through cxx and have it do anything that useful with it.

However, MoveIt does have ROS interfaces through move_group so you could write rust code using ros2-rust that communicates using ROS to MoveIt through those interfaces. You can also write a C++ application that wraps some behavior you want to express in ROS interfaces and interact through those interfaces.

closed 2026-09-26 by maspe36

tylerjw, 2026-09-26:
If I were you I'd investigate using https://github.com/ZettaScaleLabs/hiroz if you want to write Rust that works in a ROS project including using MoveIt2.

maspe36, 2026-09-26:
`rclrs` (This project) has had action support for a while now. It is more than possible to use MoveIt2 with rclrs. Additionally we support all rmw backends, not just zenoh