ros2-rust/ros2_rust #265: How to add geometry_msgs and other message types ?

whiteplatin, 2022-09-23:
Im able to use `std_msgs`
just like here https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md

but when I add `geometry_msgs` in the `cargo.toml` file ,it doesn't work.
```toml
rclrs = "*"
std_msgs = "*"
geometry_msgs = "*"
```
```
error: no matching package named `geometry_msgs` found
location searched: registry `crates-io`
required by package `turtle_rust v0.1.0 
```

nnmm, 2022-09-24:
You need to clone `geometry_msgs` into your workspace. `std_msgs` is already cloned by the `vcs import src < src/ros2_rust/ros2_rust_foxy.repos` step.

whiteplatin, 2022-09-24:
Shouldnt `geometry_msg` also be cloned alongside with `std_msgs`  ?

Since both are from `url: https://github.com/ros2/common_interfaces.git` , which is in `vcs import src < src/ros2_rust/ros2_rust_foxy.repos` 

You might as well give some instructions too

nnmm, 2022-09-24:
Oh, you're right.

Did you add `geometry_msgs` to `package.xml` of your package? Are you building with `colcon` or `cargo`?

whiteplatin, 2022-09-24:
I tried both via colcon and cargo.
And I've added geometry_msg to  package.xml

nnmm, 2022-09-24:
Ok. When you build with `colcon`, does it say that it's building `geometry_msgs`? And is `geometry_msgs` mentioned in your `.cargo/config.toml` file afterwards?

whiteplatin, 2022-09-24:
Oh, I havent noticed this before.
It does say that it finished building `Finished <<< geometry_msgs [2.01s] `  , when I did `colcon build` , but I get the `cargo build` error nonetheless.

```
Starting >>> turtle_rust
--- stderr: turtle_rust                                                                                                                                                               
error: failed to compile `turtle_rust v0.1.0 (/home/us/turtlebot3_ws/src/turtle_rust)`, intermediate artifacts can be found at `/home/us/turtlebot3_ws/src/turtle_rust/target`

Caused by:
  no matching package named `geometry_msgs` found
  location searched: registry `crates-io`
  required by package `turtle_rust v0.1.0 (/home/us/turtlebot3_ws/src/turtle_rust)`
---
Failed   <<< turtle_rust [0.83s, exited with code 101]
```

Heres the `.cargo/config.toml`:
```
[patch.crates-io.test_msgs]
path = "/home/us/turtlebot3_ws/install/test_msgs/share/test_msgs/rust"

[patch.crates-io.action_msgs]
path = "/home/us/turtlebot3_ws/install/action_msgs/share/action_msgs/rust"

[patch.crates-io.unique_identifier_msgs]
path = "/home/us/turtlebot3_ws/install/unique_identifier_msgs/share/unique_identifier_msgs/rust"

[patch.crates-io.builtin_interfaces]
path = "/home/us/turtlebot3_ws/install/builtin_interfaces/share/builtin_interfaces/rust"

[patch.crates-io.rclrs]
path = "/home/us/turtlebot3_ws/install/rclrs/share/rclrs/rust"

[patch.crates-io.rosidl_runtime_rs]
path = "/home/us/turtlebot3_ws/install/rosidl_runtime_rs/share/rosidl_runtime_rs/rust"

[patch.crates-io.turtlebot3_msgs]
path = "/home/us/turtlebot3_ws/install/turtlebot3_msgs/share/turtlebot3_msgs/rust"

[patch.crates-io.dynamixel_sdk_custom_interfaces]
path = "/home/us/turtlebot3_ws/install/dynamixel_sdk_custom_interfaces/share/dynamixel_sdk_custom_interfaces/rust"

[patch.crates-io.example_interfaces]
path = "/home/us/turtlebot3_ws/install/example_interfaces/share/example_interfaces/rust"

[patch.crates-io.std_msgs]
path = "/home/us/turtlebot3_ws/install/std_msgs/share/std_msgs/rust"

[patch.crates-io.rclrs_example_msgs]
path = "/home/us/turtlebot3_ws/install/rclrs_example_msgs/share/rclrs_example_msgs/rust"
```

`package.xml`:
```
<package format="3">
  <name>republisher_node</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="user@todo.todo">user</maintainer>
  <license>TODO: License declaration</license>

  <depend>rclrs</depend>
  <depend>std_msgs</depend>
  <depend>geometry_msgs</depend>

  <export>
    <build_type>ament_cargo</build_type>
  </export>
</package>
```

So could be a diff namespace other than

nnmm, 2022-09-24:
Very strange. Do you have a repo somewhere that I could clone to reproduce this behavior?

whiteplatin, 2022-09-24:
I do not I'll just say what I did, maybe it's something very "silly". If it can't be seen, I'll just upload it on GitHub.

1. `mkdir -p workspace/src && cd workspace`
2. `git clone https://github.com/ros2-rust/ros2_rust.git src/ros2_rust`
3. `vcs import src < src/ros2_rust/ros2_rust_foxy.repos`
4. `. /opt/ros/foxy/setup.sh`
5. `colcon build`
6. cd src
7. `~/workspace/src$  cargo new republisher_node && cd republisher_node`
8. `~/workspace/src/republisher_node/Cargo.toml`:
```
[package]
name = "republisher_node"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
rclrs = "*"
std_msgs = "*"
```
9. `colcon build`  ==> Success

10. Update `cargo.toml` by adding `geometry_msg = "*"` under `[dependencies]`:

11. `Create a new ~/workspace/src/republisher_node/package.xml`:
```
<package format="3">
  <name>republisher_node</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="user@todo.todo">user</maintainer>
  <license>TODO: License declaration</license>

  <depend>rclrs</depend>
  <depend>std_msgs</depend>
  <depend>geometry_msg</depend>

  <export>
    <build_type>ament_cargo</build_type>
  </export>
</package>
```
11. (catkin clean) : `~/workspace$ rm -r build install log`

12. `~/workspace$ colcon build`
```
--- stderr: republisher_node                                                                                                                                                                                                                                 
error: no matching package named `geometry_msg` found
location searched: registry `crates-io`
required by package `republisher_node v0.1.0
```

esteve, 2022-09-24:
> 13\. <depend>geometry_msg</depend>

The name of the package is `geometry_msgs`, in plural.

whiteplatin, 2022-09-24:
Oh yeah, its working now !
Dont know where I went wrong the first time , since I had `geometry_msgs`. 
I'll just continue with this new one. 

Thanks!

closed 2022-09-24 by whiteplatin