ros2-rust/ros2_rust #275: Create github pages with rustdoc documentation

nnmm, 2022-10-03:
docs.rs cannot build `rclrs` since it requires a ROS 2 installation, and as such it can't display the rustdocs: https://docs.rs/crate/rclrs/0.2.0

To make ros2-rust more accessible, we should host the docs ourselves, and GitHub pages are probably the easiest way. They could be updated in a CI step.

Crates that should be covered are `rclrs`, `rosidl_runtime_rs`, and perhaps generated message crates for `common_interfaces` packages (they are built in CI anyway).

esteve, 2022-10-03:
I agree, though I'd use readthedocs to host the documentation, a significant number of open source projects host their documentation there. I'd leave Github pages for non-documentation content.

michalfita, 2023-02-10:
@nnmm Have you read the https://forge.rust-lang.org/docs-rs/add-dependencies.html?

nnmm, 2023-02-10:
@michalfita Unfortunately, ROS 2 is not in the standard Ubuntu 20.04 repositories.

michalfita, 2023-02-10:
It may be tough call, but for `ros2-rust` to go mainstream docs.rs is mandatory. Either raise good cause and reason and reqest adding ROS2 to their Docker
or
Fork it with required dependencies and run own copy... Question remains open where to host it.

That BTW sounds like a feature to allow certain serious maintainers to provide own Docker image to generate documentation 🤔

maspe36, 2024-07-14:
As of https://github.com/ros2-rust/ros2_rust/pull/347, we now generate docs.rs documentation. This suggestion is now obsolete.

closed 2024-07-14 by maspe36