ros2-rust/ros2_rust #281: Make take_boxed() create messages directly on the heap

nnmm, 2022-10-08:
Currently, the implementation of `take_boxed()` will place the message on the stack during intermediate steps, even though the intent of this function is to avoid that, since large messages might overflow the stack.

This requires avoiding any `Box::new()` calls.

Nizerlak, 2022-11-07:
Hi, I'm considering contribution to this issue. However, I need some hints how to start. AFAIK [this](https://github.com/ros2-rust/ros2_rust/blob/3b6c01364f2fd068d6c2049e102e26ff26fe4277/rclrs/src/subscription.rs#L186) and [this](https://github.com/ros2-rust/ros2_rust/blob/3b6c01364f2fd068d6c2049e102e26ff26fe4277/rclrs/src/subscription.rs#L190) Box::new functions have to be replaced with another one (which? [from_raw](https://doc.rust-lang.org/std/boxed/struct.Box.html#method.from_raw)?). Also I've found the following comment in `take_boxed()`
```
        // TODO: This will still use the stack in general. Change signature of
        // from_rmw_message to allow placing the result in a Box directly.
```
How should the signature look like? I assume that it should return a pointer, but how to achieve that?
It'd be great, if you could explain that to me.
Regards

asterycs, 2022-11-07:
Hi! I've wanted to make a PR for this one as well but I've been having some issues with setting up the environment.

As far as I understand from the discussion in https://github.com/rust-lang/rust/issues/53827 the safest way would be to use `vec!` together with [`into_boxed_slice`](https://doc.rust-lang.org/std/vec/struct.Vec.html#method.into_boxed_slice) since `vec!` apparently avoids the stack allocation. I'm not sure if this still requires the signature change that is mentioned in the comment.

@Nizerlak will probably have a PR before me anyway :)

nnmm, 2022-11-12:
I've actually already looked into this a while back, so I pushed my work-in-progress branch in case one of you wants to take over: https://github.com/ros2-rust/ros2_rust/compare/main...take_boxed

Let me know if that helps. The main problem I was facing was that the `msg_idiomatic.rs.em` template code becomes very large with this change, perhaps there's a way to avoid it though.