ros2-rust/ros2_rust #291: Colcon build fails with error: "failed to load source for dependency `action_msgs`"

DS3a, 2022-11-03:
I am following the build documentation provided here (https://github.com/ros2-rust/ros2_rust/blob/main/docs/building.md#building-with-colcon)

when I run colcon build to build all the default packages, I get the following error:

```
Starting >>> rosidl_runtime_rs
Starting >>> test_interface_files
--- stderr: rosidl_runtime_rs                                                                            
error: failed to load source for dependency `action_msgs`

Caused by:
  Unable to update /home/ds3a/ros2_ws/install/action_msgs/share/action_msgs/rust

Caused by:
  failed to read `/home/ds3a/ros2_ws/install/action_msgs/share/action_msgs/rust/Cargo.toml`

Caused by:
  No such file or directory (os error 2)
---
Failed   <<< rosidl_runtime_rs [0.54s, exited with code 1]
Aborted  <<< test_interface_files [0.56s]

Summary: 0 packages finished [1.31s]
  1 package failed: rosidl_runtime_rs
  1 package aborted: test_interface_files
  1 package had stderr output: rosidl_runtime_rs
  32 packages not processed

```

nnmm, 2022-11-07:
@DS3a Have you followed the troubleshooting tips from that guide? Specifically doing a fresh build by deleting the `install*`, `build*`, and `.cargo` directories?

DS3a, 2022-11-09:
Yes I have, I even reinstalled ROS foxy. But it didn't work.

nnmm, 2022-11-12:
@esteve @jhdcs Do you have any ideas?

jhdcs, 2022-11-14:
The only thing I can think of is that this is a message library, which means we might be running into that issue where the Rust 
message generator isn't being run at the same time as the C++/Python generators (known limitation in ROS 2). To get around it when trying to build the client library itself, I would create a `ros2` folder parallel to `rclrs`/`rosidl_generator_rs`/etc. and clone the individual message repositories that were being complained about.

However, it looks like that's already been done here, apart from the outer `ros2` folder...

esteve, 2022-11-14:
@DS3a could you paste the output of the following commands? Thanks

`$ find /home/ds3a/ros2_ws`

From `/home/ds3a/ros2_ws`:

`$ colcon list`

DS3a, 2022-11-17:
I'm sorry, I was in a hurry, so I just wiped my disk and reinstalled everything. It works fine now, I doubt whether I will be able to help debug this issue.

I would like to apologize for my haste again. Should I keep this issue open for in case someone else faces the same issue? or should I close it?

nnmm, 2022-11-18:
No problem @DS3a. I'll just close the issue, people can still find it when they search for it.

closed 2022-11-18 by nnmm