ros2-rust/ros2_rust #296: Implement lifecycle nodes

jhdcs, 2022-11-29:
Some ROS 2 applications in other languages require the use of lifecycle nodes. We should work on creating/fleshing out the `ros2_version` of these, so that we can support more types of solutions.

I am unsure of how difficult this will be to port over, or if it's a good first issue for the community...

fmrico, 2023-12-30:
Hi @jhdcs

Are you currently working on this feature? In that case, may I help you?

jhdcs, 2024-01-02:
Yes I am! Though progress is admittedly slow.

I would welcome any and all help that you would like to provide! I'm currently trying to create a unit test to make sure what I've written so far is working correctly, and ironing out any bugs discovered. So far, I'm not happy with the ergonomics of the builder, though I'm drawing a blank on improving it. I've also noticed that if you try to do an invalid transition between states, a segfault occurs - that will need to be fixed as well.

All work is currently being done on my personal fork: https://github.com/jhdcs/ros2_rust/tree/lifecycle_nodes. If you see any ways to improve/tweak what's there, or any glaring faults, please feel free to either let me know or submit a PR!

berndpfrommer, 2026-09-16:
Hi, what's the progress on this? Thanks!

esteve, 2026-10-08:
@fmrico would you like to take over? Unfortuntelya @jhdcs no longer contributes to ros2-rust (hopefully that will change sometime in the future!), if you'd like to submit a PR with your changes, that'd be fantastic. Thanks!

mxgrey, 2026-10-08:
For reference, the Lifecycles slide of [this presentation](https://docs.google.com/presentation/d/1JlB2mx5hh_LxcJjPc-ognPRfxXA-5FQcrZSIi_dBPYI/edit?usp=drivesdk) outlines a recommendation of what the Lifecycles API should look like to be consistent with how async execution is done for services and actions in rclrs. Following that recommendation will allow lifecycle transitions to be idiomatic with the rest of rclrs and Rust in general, while still meeting all the expectations of the ROS 2 design docs.

I'm happy to provide guidance to anyone who wants to work on this.