ros2-rust/ros2_rust #298: Building rclrs nodes causes python nodes to break

JustSomeRandomUsername, 2023-01-23:
I am getting an 
"AttributeError: 'ParameterDescriptor' object has no attribute 'dynamic_typing'" 
when trying to start python nodes after building a rclrs node and sourcing the installation.

If I don't build and source the rust node, the python node runs fine. But after I build the rust node and source the installation, I can't sart python nodes, even "ros2 topic list" and similar commands give this error.

For the "ros2 topic" command it seems to help to call it once before sourcing the rust node, then it continues to run normally. The same does not work for my python nodes.  

I am running Ros-Humble in a podman container, with ros2_rust installed per instuctions.

maspe36, 2023-02-25:
> with ros2_rust installed per instructions

I think this may be your issue actually. The instructions in the readme have you clone the foxy repos
```
vcs import src < src/ros2_rust/ros2_rust_foxy.repos
. /opt/ros/foxy/setup.sh
```

You're likely hitting this [line](https://github.com/ros2/rclpy/blob/fe0d32af3a668c9641af1350075249238216905d/rclpy/rclpy/node.py#L470). This attribute is not present in [foxy](https://docs.ros2.org/foxy/api/rcl_interfaces/msg/ParameterDescriptor.html), seems to have been added in [galactic](https://docs.ros2.org/galactic/api/rcl_interfaces/msg/ParameterDescriptor.html). 

I was able to reproduce with the following
1. Follow the instructions as they are in the README and clone the foxy repos in the ros2_rust workspace.
2. Source'd my system humble installation
3. Source'd the ros2_rust workspace
4. Ran the py_pubsub listener example

```
sam@orange:~/rust_ws_humble$ ros2 run py_pubsub listener 
Traceback (most recent call last):

...

AttributeError: 'ParameterDescriptor' object has no attribute 'dynamic_typing'
[ros2run]: Process exited with failure 1

```

So to resolve this, make sure you clone the humble repos
```
vcs import src < src/ros2_rust/ros2_rust_humble.repos
```
Clean your workspace build and install folders, rebuild, source, and you should be good.

Guelakais, 2024-04-11:
In my setup, which is based on one of my [Docker environments](https://gitlab.com/docker5348/ros2_main_dev_container), I couldn't replicate your problem either. Would you like to tell us about your environment?
* What is your hardware architecture?
* Which Podman container do you use? Would you link it?
* What were the nodes for that caused the crash?

JustSomeRandomUsername, 2024-04-13:
Thanks for looking at this.
Ill be honest, I have not worked on  the project I had this project in, for quite a while 
and I don't have the container anymore. I would be absolutly fine closing this, It seems to be a Problem only I had, so it might have been my fault.
Sorry for forgetting about is issue.
If you want to look into this deeper I would be happy trying to repoduce my old setup.

closed 2024-04-13 by JustSomeRandomUsername