ros2-rust/ros2_rust #301: Is this project abandoned?

jodusan, 2023-02-22:
It seems last commit was in november. Is this end of the road for the closest to rclers lib?

tomkimsour, 2023-02-23:
it is not. PR and issues are being answered. I don't think maintainers of the package are full time on this so feel free to help completing this project. You can find many leads on how to do it in markdown files of this project or by looking at issues and PR.

maspe36, 2023-02-25:
I get lots of responses from the maintainers whenever I create a pull request or an issue. It is very much alive

jpace121, 2023-02-26:
4 months really isn’t a long gap.

esteve, 2023-02-26:
@jodusan thanks for your interest in `ros2-rust` ! The project is making progress (e.g. @jhdcs is working on lifecycle nodes https://github.com/ros2-rust/ros2_rust/issues/296, I myself on actions https://github.com/ros2-rust/ros2_rust/pull/295), but unfortunately neither @jhdcs , @nnmm or I work on this project full time. In any case, contributions and feedback are more than welcome. Is there a feature you'd like to see implemented in `rclrs` ? Are you planning to use `rclrs` in a project? Thanks :slightly_smiling_face:

nnmm, 2023-02-27:
Yep. For me, contributions happen in bursts depending on the amount of free time and motivation. In the list of open PRs you can also see recent activity.

closed 2023-03-01 by nnmm