ros2-rust/ros2_rust #303: rosidl_runtime_rs differences between main and crate release breaks local colcon build

texhnolyze, 2023-02-27:
Hi so I tried to get this up and running to test out integrating it with our existing [ros codebase](https://github.com/bit-bots/bitbots_meta) and ran into the following issue:
It seems the removal of libc (https://github.com/ros2-rust/ros2_rust/pull/284) and thus change of the `get_type_support` to return `*const std::os::raw::c_void` leads to the build breaking:
[image]

After some investigation this is only due to the latest release 0.3.1 installed from [crates.io](https://docs.rs/crate/rosidl_runtime_rs/0.3.1/source/src/traits.rs) not having this change yet.
As such this is only an issue during local development on the main branch and might just be due to my limited knowledge of rust/cargo.
However it was very confusing for me that `rosidl_runtime_rs` is downloaded and is used instead of the local one built with `colcon build`.
The [section](https://github.com/ros2-rust/ros2_rust/blob/main/docs/building.md#learning-by-doing) in the documentation does not seem up to date. As I understand the `rosidl_runtime_rs` crate did not use to exist?
By changing `Cargo.toml` of `rclrs` to use `rosidl_runtime_rs = { path = "../rosidl_runtime_rs" }` I'm at least able to build with cargo.

What would be the correct way to make a `colcon build` have `rclrs` use the locally built `rosidl_runtime_rs` instead of using the crate?

nnmm, 2023-03-01:
Thanks for the report! I'm trying to understand how the two versions got mixed – could you tell me how you depend on `rclrs` in your project (i.e. do you have a checkout of the repo, or through crates.io? And how is rclrs referenced in your Cargo.toml?)? And what is listed for `rclrs` and `rosidl_runtime_rs` in `~/.cargo/config.toml`?

texhnolyze, 2023-03-03:
Ok while typing the answer I realized that you were talking about  `~/.cargo/config.toml`.
As colcon creates a bunch of files/folders I have its own folder as a workspace under `~/colcon_ws` resulting in the patch configuration for the path of `rosidl_runtime_rs` ending up in `~/colcon_ws/.cargo/config.toml` and not being applied.
Copying the file to the correct folder in the homedir fixes the issue.
I suppose in this case it is more an issue with `colcon-ros-cargo`?
Looking at the code it the `.cargo/config.toml` is place in the working directory https://github.com/colcon/colcon-ros-cargo/blob/main/colcon_ros_cargo/task/ament_cargo/build.py#L100.
So I guess it not working is maybe due to the execution directory of `cargo` in the `colcon build`, as according to the [docs](https://doc.rust-lang.org/cargo/reference/config.html) if it is executed in the working dir the configuration should be applied.
Unless passing `--manifest-path` or `--target-dir` [here](https://github.com/colcon/colcon-ros-cargo/blob/main/colcon_ros_cargo/task/ament_cargo/build.py#L83,L90) change that.

Anyway it seems to not be an issue with `ros2_rust`. 
@nnmm I saw that you are also maintaining `colcon-ros-cargo`. 
I'm happy to open an issue there and do a PR if you could point me in the right direction.

texhnolyze, 2023-03-03:
After a quick search, this might even be just a bug in `cargo` itself https://github.com/rust-lang/cargo/issues/2930