ros2-rust/ros2_rust #314: Getting an Result::unwrap() Error when launching a launch file.

juliamp22, 2023-06-14:
Hello,
I just want to make a launch file to run a rust node.
When I run this launch file: 

from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package = 'teleop_twist_keyboard',
            executable = 'teleop_twist_keyboard',
            output = 'screen'
        )
])

It launch with failure giving me the following error: 

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Os { code: 25, kind: Uncategorized, message: "Inappropriate ioctl for device" }', teleop_twist_keyboard/src/teleop_twist_keyboard.rs:81:47
[teleop_twist_keyboard-1] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[ERROR] [teleop_twist_keyboard-1]: process has died [pid 72332, exit code 101, cmd '/home/julia/workspace_rust/install/teleop_twist_keyboard/lib/teleop_twist_keyboard/teleop_twist_keyboard --ros-args -r __node:=teleop'].

```
If I run the node with ros2 run <pkg> <node_name> it works perfectly.

I have seen that it also appeared in issue[85](https://github.com/ros2-rust/ros2_rust/issues/85). 

I have tried to check the cargo-ament-build version, but I am not able to find it.

Thank you very much for your help,

Júlia Marsal

esteve, 2023-06-14:
@juliamp22 thanks for trying out `ros2-rust` :slightly_smiling_face: You can see the version of the `crates` you've installed with:

```sh
$ cargo install --list
```

That should report the version of `cargo-ament-build` that you have installed.

As for the error, it could be because `launch` captures the `stdin` (https://github.com/juliamp22/rust_common_pkgs/blob/master/teleop_twist_keyboard/src/teleop_twist_keyboard.rs#L81), but I could be wrong. The Python package uses different attributes for `termios`, perhaps that makes a difference.

juliamp22, 2023-06-14:
Hello @esteve  thank you very much for your quick response. Yes, you are totally right that in the Python package uses different attributes for the `termios`. I'll have a deep look. You can proceed with closing the current issue.

closed 2024-03-20 by maspe36