ros2-rust/ros2_rust #320: `extern` block uses type `u128`, which is not FFI-safe

maspe36, 2023-07-09:
Seeing this new warning now with rust 1.63.0. This is causing CI to fail 

```
warning: `extern` block uses type `u128`, which is not FFI-safe
    --> /home/sam/rust_ws/src/ros2_rust/rclrs_tests/target/debug/build/rclrs-8555e18c0c9bb180/out/rcl_bindings_generated.rs:6343:31
     |
6343 |         dynamic_type_builder: *mut rosidl_dynamic_typesupport_dynamic_type_builder_t,
     |                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not FFI-safe
     |
     = note: 128-bit integers don't currently have a known stable ABI

```

Seems to be coming from auto-generated idl code. Investigate and hopefully resolve

esteve, 2023-07-12:
It's interesting that this only happens with `rolling`. As I see it, we have two options:

 - disable the dynamic message feature (`dyn_msg`) from the `clippy` check
 - disable the warning about `u128` (via `improper-ctypes`) from the `clippy` check

I believe that disabling the check from `clippy` should be fine in our case, as we're not exposing this API.

@maspe36 what do you think?

maspe36, 2023-07-13:
This is also happening with `iron` in my PR to add support for it. I'm trying to hunt down the root cause and depending on that it may indeed make sense to disable the warning or dynamic message feature from clippy.

The first build I found where this was failing the rolling [build #974](https://github.com/ros2-rust/ros2_rust/actions/runs/4998294792/jobs/8953551442). 

Current theory is that this is related to the version uptick in rosidl_generator_c from 3.3.1 to 4.1.1 as that is a major version change between builds #974 and [#973](https://github.com/ros2-rust/ros2_rust/actions/runs/4986815835/jobs/8927975778). I'll keep digging

maspe36, 2023-07-13:
Looks like this is cause by the new [Runtime Interface Reflection feature (#215)](https://github.com/ros2/rmw_implementation/pull/215) in the [rmw_implementation](https://github.com/ros2/rmw_implementation) package. 

However, the original type that is ultimately by translated by bindgen(?) to a u128 is defined [here](https://github.com/ros2/rosidl_dynamic_typesupport/blob/rolling/include/rosidl_dynamic_typesupport/api/dynamic_data.h#L47) in the newly stabilized in Iron, [rosidl_dynamic_typesupport](https://github.com/ros2/rosidl_dynamic_typesupport/tree/rolling) package.

It seems like the `dyn_msg` feature is a custom feature we use and frankly I don't know much about cargo features. I think the best course would be to disable clippy lints for the auto generated bindgen code. 

But... Isn't that what we're doing [here](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/rcl_bindings.rs#L6)? Safe to say I'm a bit lost

closed 2023-07-15 by esteve

dev-ardi, 2024-05-13:
u128 is already FFI safe: https://blog.rust-lang.org/2024/03/30/i128-layout-update.html

maspe36, 2024-05-13:
Yup, this is now part of stable Rust as of May 2nd. To take advantage we will need to upgrade from 1.74 to 1.78.

Issue for this upgrade created here https://github.com/ros2-rust/ros2_rust/issues/398

PR #322 (2023-07-15): Disable clippy warnings for u128 types
Fixes #320