ros2-rust/ros2_rust #336: Docs Fail to Generate on Docs.rs

Carter12s, 2023-10-24:
https://docs.rs/crate/rclrs/0.3.1

[image]

From checking the build log its obvious they failed because no ROS_DISTRO was installed, but I think it should be possible to work around this with feature flags and get documentation to generate without needed ROS.

Docs.rs is pretty central in the Rust community so would be good to support.

esteve, 2023-10-27:
That's a really good idea. From what I gathered, if we add a feature flag that avoid dependening on `ROS_DISTRO` and add it to `[package.metadata.docs.rs]` (https://docs.rs/about/metadata), we could make it work, right? What do you think @Carter12s ?

Carter12s, 2023-10-27:
Yup that is exactly what I was thinking. I'm not sure how extensive that feature flag is going to need to be and unfortunately there isn't a great way to test other than releasing versions and seeing if doc.rs picks them up.

When I was getting this figured out for roslibrust I had to release and then yank several versions just to test if docs.rs would parse them correctly.

esteve, 2023-10-27:
Hmmm, it could be tricky, because we depend on `bindgen` for a significant part of the API, which depends on having a ROS distro installed. Anyway, I'm sure we can figure out a way.

PR #347 (2023-11-24): Allow rustdoc to run without a ROS distribution
Fixes #336
closed 2023-11-24 by esteve