ros2-rust/ros2_rust #340: 0.4 Release Tracking Issue

maspe36, 2023-10-28:
This issue tracks the 0.4 ros2_rust release.

Merged PRs
- Remove libc dependencies ([#284](https://github.com/ros2-rust/ros2_rust/pull/284))
- Basic functionality for loading introspection type support libraries ([#279](https://github.com/ros2-rust/ros2_rust/pull/279))
- Vendor rcl_interfaces as a private module into rclrs ([#288](https://github.com/ros2-rust/ros2_rust/pull/288))
- Make constructor methods for pub/sub/etc. take rcl_node_t mutex ([#290](https://github.com/ros2-rust/ros2_rust/pull/290))
- Extend string types ([#293](https://github.com/ros2-rust/ros2_rust/pull/293))
- Allow rolling job to fail. Force install of known to work version of ([#302](https://github.com/ros2-rust/ros2_rust/pull/302))
- Removed support for the EOL galactic distro ([#306](https://github.com/ros2-rust/ros2_rust/pull/306))
- Improve efficiency of deserializing strings ([#300](https://github.com/ros2-rust/ros2_rust/pull/300))
- Fix hardcoded rmw_gid_t length ([#309](https://github.com/ros2-rust/ros2_rust/pull/309))
- Use ros2-testing debs for rolling ([#308](https://github.com/ros2-rust/ros2_rust/pull/308))
- Swapped usage of rosidl_cmake over to the new rosidl_pycommon. ([#297](https://github.com/ros2-rust/ros2_rust/pull/297))
- Remove Foxy from the list of supported ROS 2 distributions ([#318](https://github.com/ros2-rust/ros2_rust/pull/318))
- Allow usage of different ROS distributions in Dockerfile ([#313](https://github.com/ros2-rust/ros2_rust/pull/313))
- Disable clippy warnings for u128 types ([#322](https://github.com/ros2-rust/ros2_rust/pull/322))
- Update Rust toolchain to 1.71.0 and remove pydocstyle workaround ([#321](https://github.com/ros2-rust/ros2_rust/pull/321))
- Add support for Iron Irwini ([#319](https://github.com/ros2-rust/ros2_rust/pull/319))
- Bindgen usage improvements ([#323](https://github.com/ros2-rust/ros2_rust/pull/323))
- Return Arc from the create_node function to match other create_X functions ([#294](https://github.com/ros2-rust/ros2_rust/pull/294)) 
- Refactor spin and spin_once to mimic rclcpp's Executor API ([#324](https://github.com/ros2-rust/ros2_rust/pull/324))
- add serde big array support ([#328](https://github.com/ros2-rust/ros2_rust/pull/328))
- Fixed an issue where cpu usage remained at 100% ([#330](https://github.com/ros2-rust/ros2_rust/pull/330))

PRs yet to be merged

- [ ]  Add time source and clock API to nodes ([#325](https://github.com/ros2-rust/ros2_rust/pull/325))

esteve, 2023-10-31:
@maspe36 thanks for collecting all this info. Perhaps it'd be useful to split this into features, updates, fixes, etc. what do you think?

maspe36, 2023-11-04:
Gave it a shot, let me know what you think

esteve, 2023-11-06:
@maspe36 awesome, thanks!

esteve, 2023-11-06:
@maspe36 I've updated the ticket now that https://github.com/ros2-rust/ros2_rust/pull/325 is merged. I'll tag and create new release tomorrow.

esteve, 2023-11-07:
Released!! https://crates.io/crates/rclrs/0.4.0 :partying_face:

closed 2023-11-07 by esteve