ros2-rust/ros2_rust #351: A small tool for creating ros2 rust packages is available.

Guelakais, 2023-11-20:
I found it really tedious to build the ros2 rust packages myself every time. So over the weekend I've written a tool to do this automatically. Is this interesting for you?
[ros2_rust_packaging_tool](https://gitlab.com/rust_projects3505446/ros2_rust_packaging_tool/-/tree/ROS2_package_creation?ref_type=heads)

esteve, 2023-11-21:
@Guelakais this is really cool, thanks!

Have you checked https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html ? The traditional way of creating a package in ROS 2 is via the `ros2` tool (i.e. `ros2 pkg create ...`), unfortunately it only supports packages that use the `ament_cmake` or `ament_python` build types, but it'd be really useful if it supported other build types (i.e. `ament_cargo`).

Perhaps you could reach out to the developers of the `ros2` tool and find a way to make it extensible to any build type. The repository for the `ros2` tool is https://github.com/ros2/ros2cli/tree/rolling and you can the code for `ros2 pkg create` at https://github.com/ros2/ros2cli/blob/rolling/ros2pkg/ros2pkg/verb/create.py

Guelakais, 2023-11-21:
I knew the tutorial, like the ros2 command line tool and had some looks at the source code. I'm not sure if it will be expandable in the near future to support ament-cargo or other new build types.
My tool isn't intended to replace the ros2 command line tool. It's more about getting more experience with Rust by building something I want to use myself.

Guelakais, 2024-04-22:
I consider this problem to be solved.

closed 2024-04-22 by Guelakais